cesanta/mongoose-os #103: Can't build the latest changes - unable to ping registry endpoint https://docker.cesanta.com:5000/v0/

lukevenediger, 2016-03-17:
Hi Guys,

I'm trying to build the latest changes (https://github.com/cesanta/smart.js/commit/29796572d6aa14fd62969e6f6f26dd3a15e03108) and I get this error when running 'make'.

Any idea why this is failing?

```
esp8266 $ make
make: Entering directory `<redacted>/smartjs/platforms/esp8266'
Unable to find image 'docker.cesanta.com:5000/esp8266-build-oss:1.5.2-r2' locally
docker: Error response from daemon: unable to ping registry endpoint https://docker.cesanta.com:5000/v0/
v2 ping attempt failed with error: Get https://docker.cesanta.com:5000/v2/: dial tcp: lookup docker.cesanta.com on 192.168.1.1:53: read udp 10.0.2.15:54842->192.168.1.1:53: i/o timeout
 v1 ping attempt failed with error: Get https://docker.cesanta.com:5000/v1/_ping: dial tcp: lookup docker.cesanta.com on 192.168.1.1:53: read udp 10.0.2.15:35353->192.168.1.1:53: i/o timeout.
See 'docker run --help'.
make: *** [all] Error 125
make: Leaving directory `<redacted>/smartjs/platforms/esp8266'
```

Thanks,
Luke

mmikulicic, 2016-03-17:
Hi,

```
lookup docker.cesanta.com on 192.168.1.1:53: read udp 10.0.2.15:54842->192.168.1.1:53: i/o timeout
```

this looks like a DNS problem on your side. Are you using docker-machine on osx ? If, yes try to restart the VM.

lukevenediger, 2016-03-17:
Ok thanks @mmikulicic I'll try this now.

lukevenediger, 2016-03-17:
That fixed it - thanks @mmikulicic!

closed 2016-03-17 by lukevenediger