cesanta/mongoose-os #105: heap integrity broken: block links don't match: 5613 -> 424, but 424 -> 16383

realzeus, 2016-03-21:
I have got the error message "heap integrity broken: block links don't match: 5613 -> 424, but 424 -> 16383" when I tried to execute a generated JS.
Dump is attached, source is attached.

The goal was to use Scala as a development language due to its advantages:
- strictly typed language
- strictly typed integration with server (no more problems on renaming field of json transfer object)
- and many other.

"heap integrity broken" is not "out of memory", so I still hope that it can be done.
It would be awesome!

[js.zip](https://github.com/cesanta/smart.js/files/183090/js.zip) - generated JS and optimized JS
[dump.zip](https://github.com/cesanta/smart.js/files/183086/dump.zip)
[fullSource.zip](https://github.com/cesanta/smart.js/files/183087/fullSource.zip) - see `fullSource\src\main\scala\z\Main.scala`; start by executing `activator run` or generate JS `activator fastOptJS` / `activator fullOptJS`

PS: Firmware - nightly build from 20th March, Hardware ESP-01

mmikulicic, 2016-03-22:
It looks like an out of memory. Relevant log:

```
E:M 3070 (385 blocks)
E:M 3069 (385 blocks)
E:M 3070 (385 blocks)
```

This is an out of memory condition, followed by a heap corruption likely caused by a mismanaged handling of the out of memory condition. 

There is hope though :-) 

Do you precompile the JS with `v7 -c` o `smartjs -c` (the smartjs posix port) ?
Doing so will allow Smart.js to keep most of code and read-only data on the flash filesystem.

If not, keep in mind that the device has about 30KB of free heap and you're trying to parse and load in memory a file whose bytecode is about 15KB, including strings etc. Please give precompilation a try.

In order to pre-compile, build https://github.com/cesanta/v7 or just use the v7 binary in `smartjs/platforms/esp8266/build/v7`. 

Furthermore we currently have a lot of leeway for optimizations, for example we currently grow the string heap with realloc which is suboptimal to say the least (having a contiguous heap does simplify the compacting collector though).

realzeus, 2016-03-22:
Thanks for quick response. I will compile V7 later and try to precompile the JS.

What is the output of `v7 -c`? Is it a `.js` file, which is not readable?
What should I do with it and how can I execute it? `File.eval()`?

mmikulicic, 2016-03-22:
the output of `v7 -c`  outputs a binary bytecode file, and `File.eval()` will detect the format and run it regardless of the extension; feel free to use the `.js` extension or `.jsc` if you prefer. 
System `.js` files shipped in the FW filesystem such as `sys_init.js` are actually precompiled binary files.

realzeus, 2016-03-28:
There are 2 problems:
1. I have a file `print.js` with following code:
```
(function() {
  print("Hello World");
}).apply(this);
```
I execute `v7 -e 'File.eval("print.js");'` and it works correctly.
When I execute `v7 -c print.js >print.jsc` and then `v7 -e 'File.eval("print.jsc");'`, I see this error: `v7: ../../../v7/v7.c:14712: b_exec: Assertion 'fr == 0' failed.`

2. I have managed to convert Scala.js file into JavaScript, which was successfully executed by V7 on a Debian. Source JS file has 19457 bytes. `v7 -c` completes without errors and produces only 827 bytes. Compiled code does not contain referenced strings from original code.

The code is in attached `step2.zip`

closed 2016-05-08 by realzeus