cesanta/mongoose-os #106: DHT22/DHT11 Support

realzeus, 2016-03-21:
Please recover DHT22 and DHT11 support (temperature sensors).

It was initially available in [platforms/esp8266/user/dht11.c](https://github.com/cesanta/smart.js/blob/d37ee43f47adacf5571a445807713b38d394619f/platforms/esp8266/user/dht11.c)

There is an example in internet how to use DHT, but it doesn't work anymore.

alashkin, 2016-04-15:
Sorry, but  DHT11 support was discontinued.

vinnitu, 2016-05-18:
Why? I need too

rojer, 2016-05-18:
it was disabled and APIs had changed. but it's a popular sensor, i think we should re-add it back.

realzeus, 2016-05-19:
I have implemented support for DHT22 in beta4.
Hope, it will be usable.

This implementation is based on the Arduino library and DHT22 specification.
[dht22_for_beta4.zip](https://github.com/cesanta/mongoose-iot/files/271871/dht22_for_beta4.zip)

vinnitu, 2016-05-19:
what about dht11?

realzeus, 2016-05-23:
If mongoose-iot developers will confirm that they will include this code in future releases, I will extend my code to support DHT11 too.

realzeus, 2016-05-23:
By the way, I have decided to implement DHT22 because newer sensors AM2320 supports this old protocol and it's the default protocol.

eduard-sukharev, 2016-08-06:
@realzeus thanks for sharing this. As far as i can see there's some code for v7 - does it mean that I can use it with JavaScript?
Can you tell where to put the archive contents and show JavaScript example usage?

realzeus, 2016-08-08:
Sorry, still learning how to do Git Pull Requests... Shame on me! But I'm working on it.

In order to integrate this code please do following:
1 get Mongoose IoT Firmware Source
2 build Mongoose IoT Firmware Source (without my code)
3 find in Mongoose IoT Firmware Source file named `v7_esp.c` and `v7_esp_features.h`
4 copy `dht22.h` and `dht22.c` files into the same directory
5 merge my `v7_esp.c` and `v7_esp_features.h` into corresponding files (this should be trivial)
6 build the code

I have connected DHT22 to PIN 2, so I use it as `DHT22.read(2)`.
Here is more code, how I use it:

    function sendData() {
      print(GC.stat());
      var data = readData();
      if(data != null) {
        var requestParams = {
          hostname: '...',
          port: 80,
          path: '/data/' + Sys.ro_vars.mac_address,
          method: 'POST',
          headers: { 'Content-Type': 'text/json' }
        };
        Http.request(requestParams, function() {;}).setTimeout(2000, function() { this.abort(); }).end(data);
      }
    }
    function readData(){
      var sensorData = DHT22.read(2);
      if(sensorData == null) return null;
      var deviceId = Sys.ro_vars.mac_address;
      return '{"deviceId": "' + deviceId + '", ' +
          '"data": {"temperature": ' + sensorData.temperature + ', "humidity": ' + sensorData.humidity + '}}';
    }
    setInterval(sendData, 60000);

eduard-sukharev, 2016-08-10:
@realzeus
In order to make a PR you have to:
0. Fork this repo and clone your fork to your local development machine (`git clone <repo url>`)
1. Create new git branch with `git checkout -b add_dht`
2. Make your changes to the code
3. Prepare your changes for commit (add changes to *stage* and add new files too)
4. Commit your changes with `git commit -m "commit message here"`
5. Push your local branch to your remote original repository (your fork on github) with `git push --set-upstream origin add_dht`
6. Go to github and it will promptly ask you if you would like to make a pull request

apriantama, 2017-03-08:
hi @realzeus 

could you please update your steps, with the current repo https://github.com/cesanta/mongoose-os
I couldn't find Mongoose IoT Firmware Source anymore.

Thanks

r1kkman, 2017-03-15:
Hi there! Can you help me out adding Bosch BMP280 driver into mongoose os?
Thanks!

cpq, 2017-05-09:
https://github.com/cesanta/mongoose-os/tree/master/fw/examples/c_arduino_dht

closed 2017-05-09 by cpq