cesanta/mongoose-os #107: setTimeout pauses at least for 1 second

realzeus, 2016-03-22:
`setTimeout(blink, 100);` calls `blink` exactly after 1 second (instead of 0.1)
`setTimeout(blink, 1900);` works correctly (exactly after 1.9 seconds)

```
var pin = 2;
GPIO.setmode(pin, 2, 0);

var level = false;
function blink() {
  GPIO.write(pin, level);
  level = !level;
  if(level) {
    console.log(Sys.time(), " pin:", pin, ", level:", level, ", pause:", 100);
    setTimeout(blink, 100);
  } else {
    console.log(Sys.time(), " pin:", pin, ", level:", level, ", pause:", 1900);
    setTimeout(blink, 1900);
  }
}
setTimeout(blink, 100);
```

mmikulicic, 2016-03-22:
One-liner to reproduce it:

```js
function tick() { print("tick"); setTimeout(tick, 100); }; tick();
```

It's a bug and will be fixed. 
`setInterval` works though

---

Confirmed regression of 9b08114c776d3ea61104da5286dd46bc95993334.

Additional info:
```js
function tick() { print("tick"); setTimeout(tick, 100); }; tick();
setInterval(function() { print(".") }, 100); // this will cause `tick` to be also printed out every 100ms
```

rojer, 2016-03-22:
we need to trigger a mongoose poll when scheduling a timer.

rojer, 2016-03-22:
fixed by 13780e6e91edb8ca2b39269d2efc70d4b67d561d

closed 2016-03-22 by rojer