cesanta/mongoose-os #112: WebDAV does not work

kroshilin, 2016-04-03:
I flashed ESP-01 with Beta3. Everything is ok, but on Windows 10 I can't establish connection using WebDAV. Error is "Wrong folder name".

kroshilin, 2016-04-04:
However, I can send files to device using curl -T filename.txt http://192.168.1.121/filename.txt

4713, 2016-04-14:
WebDAV is also not functioning with davfs2 on Raspberry or tools like carotdav on Windows.
Both tools seem to mount the device but read/write is not possible.

cpq, 2016-04-14:
cc @alashkin 

After spending much time fixing WebDAV, we decided to discontinue WebDAV feature.
Using WebDAV would be great, but making it work takes a LOT of effort.
We will resort to HTTP POST(curl), and also we're working on the online editor that will make things easy.

@alashkin , please remove WebDAV from the documentation, and from the source code, and close this bug.

alashkin, 2016-04-14:
Removed in https://github.com/cesanta/smart.js/commit/5d2953a6f412bd9663fec8ce0fbcfd46b7715583

closed 2016-04-14 by alashkin