cesanta/mongoose-os #114: Possible missing file after

lukevenediger, 2016-04-06:
Hi,

I've just pulled the latest changes and am getting an error due to "smartjs/src/sj_utils.h" being missing. The build fails on sj_mongoose_ws_client.c. Looking at the history it seems to be related to this change: https://github.com/cesanta/smart.js/commit/484deba7d99ad38f3be73c6bf7d4093383a24afc

Any idea if the file is missing or if I'm doing something wrong?

Thanks!
Luke

Here's the full build log:

machine:esp8266 lukevenediger$ make
make: Entering directory `/Users/lukevenediger/iot-src/iot-smartjs/smartjs/platforms/esp8266'
make: Entering directory `/cesanta/v7'
make: Nothing to be done for `v7.c'.
make: Leaving directory `/cesanta/v7'
make: Entering directory `/cesanta/krypton'
make: Nothing to be done for `krypton.c'.
make: Leaving directory `/cesanta/krypton'
make: Entering directory `/cesanta/mongoose'
make: Nothing to be done for `mongoose.c'.
make: Nothing to be done for `mongoose.h'.
make: Leaving directory `/cesanta/mongoose'
make: Entering directory `/cesanta/smartjs/platforms/esp8266'
  MKDIR firmware
  MKDIR /cesanta/smartjs/platforms/esp8266/.build
CC esptool2.c
gcc -O2 -Wall -c esptool2.c -o /cesanta/smartjs/platforms/esp8266/.build/esptool2.o
CC esptool2_elf.c
gcc -O2 -Wall -c esptool2_elf.c -o /cesanta/smartjs/platforms/esp8266/.build/esptool2_elf.o
LD /cesanta/smartjs/platforms/esp8266/.build/esptool2
gcc -o /cesanta/smartjs/platforms/esp8266/.build/esptool2 /cesanta/smartjs/platforms/esp8266/.build/esptool2.o /cesanta/smartjs/platforms/esp8266/.build/esptool2_elf.o
CC rboot-stage2a.c
LD /cesanta/smartjs/platforms/esp8266/.build/rboot-stage2a.elf
E2 /cesanta/smartjs/platforms/esp8266/.build/rboot-hex2a.h
CC rboot.c
LD /cesanta/smartjs/platforms/esp8266/.build/rboot.elf
E2 /cesanta/smartjs/platforms/esp8266/.build/rboot.bin
  GEN   /cesanta/smartjs/platforms/esp8266/ld/smartjs.ld (0xcaff0 @ 0x40211010) -> /cesanta/smartjs/platforms/esp8266/.build/0x11000.ld
  GEN   /cesanta/smartjs/platforms/esp8266/.build/sys_config.c
make -C /cesanta/smartjs/platforms/posix V7_FEATURES='-DV7_ENABLE__File__require=1 -DV7_BUILD_PROFILE=1 -DV7_ENABLE_FILE -DV7_DISABLE_AST_TAG_NAMES -DV7_ENABLE__Math -DV7_ENABLE__Math__max -DV7_ENABLE__Math__min -DV7_ENABLE__Math__random -DV7_ENABLE__Math__round -DV7_ENABLE__Function__bind -DV7_ENABLE__Function__call -DV7_ENABLE__File__list=1 -DV7_ENABLE__Memory__stats -DV7_ENABLE__StackTrace=0 -DV7_ENABLE__Object__getPrototypeOf -DV7_ENABLE__Object__isPrototypeOf -DV7_ENABLE__Object__keys -DV7_ENABLE__Object__isExtensible -DV7_EXPOSE_PRIVATE -DV7_ENABLE_ENTITY_IDS  -DV7_MAIN' BIN_DIR=/cesanta/smartjs/platforms/esp8266/.build/posix_sjs_bin BUILD_DIR=/cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build CFLAGS_EXTRA='-DV7_FREEZE'
make[1]: Entering directory `/cesanta/smartjs/platforms/posix'
  MKDIR /cesanta/smartjs/platforms/esp8266/.build/posix_sjs_bin
  MKDIR /cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build
make[2]: Entering directory `/cesanta/v7'
make[2]: Nothing to be done for `v7.c'.
make[2]: Leaving directory `/cesanta/v7'
  GEN   /cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build/sys_config.c
  CC    lnx_adc.c -> /cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build/lnx_adc.o
  CC    lnx_debug.c -> /cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build/lnx_debug.o
  CC    lnx_gpio.c -> /cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build/lnx_gpio.o
  CC    lnx_i2c.c -> /cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build/lnx_i2c.o
  CC    lnx_pwm.c -> /cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build/lnx_pwm.o
  CC    lnx_spi.c -> /cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build/lnx_spi.o
  CC    lnx_wifi.c -> /cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build/lnx_wifi.o
  CC    main.c -> /cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build/main.o
  CC    posix_hal.c -> /cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build/posix_hal.o
  CC    posix_uart.c -> /cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build/posix_uart.o
  CC    smartjs.c -> /cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build/smartjs.o
  CC    ../../../v7/v7.c -> /cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build/v7.o
  CC    ../../src/sj_v7_ext.c -> /cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build/sj_v7_ext.o
  CC    ../../../mongoose/mongoose.c -> /cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build/mongoose.o
  CC    ../../src/sj_i2c_js.c -> /cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build/sj_i2c_js.o
  CC    ../../src/sj_spi_js.c -> /cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build/sj_spi_js.o
  CC    ../../src/sj_mongoose.c -> /cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build/sj_mongoose.o
  CC    ../../src/sj_mongoose_ws_client.c -> /cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build/sj_mongoose_ws_client.o
../../src/sj_mongoose_ws_client.c:16:34: fatal error: smartjs/src/sj_utils.h: No such file or directory
 #include "smartjs/src/sj_utils.h"
                                  ^
compilation terminated.
make[1]: *** [/cesanta/smartjs/platforms/esp8266/.build/posix_sjs_build/sj_mongoose_ws_client.o] Error 1
make[1]: Leaving directory `/cesanta/smartjs/platforms/posix'
make: *** [/cesanta/smartjs/platforms/esp8266/.build/posix_sjs_bin/smartjs] Error 2
make: Leaving directory `/cesanta/smartjs/platforms/esp8266'
make: *** [all] Error 2
make: Leaving directory `/Users/lukevenediger/iot-src/iot-smartjs/smartjs/platforms/esp8266'

javatai, 2016-04-08:
I have same situation like  you have.

mmikulicic, 2016-04-08:
Fixed

closed 2016-04-08 by mmikulicic

lukevenediger, 2016-04-08:
Thanks @mmikulicic!