cesanta/mongoose-os #116: The ESP8266 chip ID is not made available through the API

carterw, 2016-04-06:
http://esp8266-re.foogod.com/wiki/System_get_chip_id_%28IoT_RTOS_SDK_0.9.9%29

uint32 system_get_chip_id(void)

Returns the unique chip ID burned into the ESP8266 at the factory.

alashkin, 2016-04-12:
@carterw 

Hi! 
Could you share use case for such function?

lukevenediger, 2016-04-12:
@alashkin I have a use case to include a unique ID when the chip authenticates. I could use the MAC address but a chip ID is harder to spoof.

carterw, 2016-04-12:
If you have a lot of devices deployed for a variety of purposes you need a way to uniquely identify each of them so you can know what they are supposed to be doing and perhaps where they are.

mmikulicic, 2016-04-12:
Do you know whether there are enough guarantees that those IDs are actually unique ?
In alternative, you could bake in a unique identifier while flashing the FW with your app.

carterw, 2016-04-12:
"unique chip ID burned into the ESP8266 at the factory" sounds pretty unique to me and immutable. I don't intend to build and flash firmware if I can avoid it.

mmikulicic, 2016-04-12:
I don't see any reason why not to expose this API. I was just wondering if anybody had had any direct experience or accounts of any clashes due to operational issues in the fabrication process.

lukevenediger, 2016-04-13:
@mmikulicic I like the idea of having both as an option. Having an easy way to set an ID on the chip means that I can set the ID before flashing, and I don't need to query the device before adding it to a device registry database. On the other hand, I like what @carterw says about not having to stamp an ID into the chip - that is pretty convenient and probably harder to spoof.

rojer, 2016-04-13:
@lukevenediger chip ID is harder to spoof

that would be true, if chip id weren't last 3 octets of the mac address.
the only difference is that mac address can be changed through api and chip id cannot.

mmikulicic, 2016-04-13:
The mac address can be changed by software, but that will likely happen after our `Sys.ro_vars.mac_address` gets initialized, so at this moment `Sys.ro_vars.mac_address` is for all intents and purposes immutable and offers the same uniqueness guarantees as the chip ID (given its contained in it).

So, I guess the problem boils down to: what if you choose the way you report the mac address?

My take on this is that `Sys.ro_vars` should expose something that can be changed at runtime; if we every expose an API to change the MAC address, we'll also expose an API to retrieve that soft MAC address and leave the factory mac address in `Sys.ro_vars`.

@lukevenediger: would it be ok for you to use the `Sys.ro.mac_address` as unique identifier given the guarantee that it will always contain the factory mac address and not the interface (soft) mac address?

carterw, 2016-04-13:
Personally, seeing as how the chip ID is in the MAC address, I am fine with using that.

alashkin, 2016-04-14:
Ok, in this case I'm closing this issue

closed 2016-04-14 by alashkin

mmikulicic, 2016-04-14:
Ok, closing the issue.

If the situation changes we can always add reopen this topic when a good use case arises.

To summarize: current rationale for keeping `Sys.ro_vars.mac_address` as source of unique identifier:

1. Smartjs is multiplatform. Not all chips have unique identifiers. All networked platforms have a mac address (well, technically not all; you don't need a mac address for transports such as SLIP)
2. we care about simplicity and we prefer smaller APIs wherever possible
3. `Sys.ro_vars.mac_address` is meant to capture the hw mac address that identifies this device as opposed to provide access to the current mac address of a given interface. A device (e.g. a gateway device) could have multiple interfaces; this variable is useless to determine mac addresses in the context of networking; it's useful by definition only to provide a source of unique identity.

suculent, 2017-08-20:
```
void get_mac_address (uint8_t * mac_uin8t, char * mac_str)
{
    device_get_mac_address(mac_uin8t);
    *mac_uin8t=0x5C;                //Originally informs 5E (AP MAC ADDRESS)
    *(mac_uin8t+1)=0xCF;                            //Not really necessary
    *(mac_uin8t+2)=0x7F;                            //Not really necessary
    sprintf(mac_str, "%02x%02x%02x%02x%02x%02x\n", mac_uin8t[0]&0xFF,mac_uin8t[1]&0xFF,mac_uin8t[2]&0xFF,mac_uin8t[3]&0xFF,mac_uin8t[4]&0xFF,mac_uin8t[5]&0xFF);
    puts(mac_string);
}
```

I am on ESP32 and this crashes, also Sys.ro_vars.mac_address in mJS is useless (ro_vars do not exist). Any help would be appreaciated. https://github.com/suculent/thinx-firmware-esp8266-mos.git

It's such a stupid showstopper not being able to get MAC address for so long already...

cpq, 2017-08-21:
See https://github.com/cesanta/mongoose-os/issues/221#issuecomment-323605005