cesanta/mongoose-os #117: [Feature request] OweWire bus support

amid0, 2016-04-07:
Hello.
I wounder is the any plans to support Dallas OneWire bus in Smart JS?
There are many useful sensors and IC's with one wire bus.
Also this bus called iButton bus.
First of all for me interesting popular temperature sensor DS18B20. By the way NodeJS adaptation for esp8266 already support this sensor )).
Here is few links:
https://en.wikipedia.org/wiki/1-Wire
https://www.maximintegrated.com/en/app-notes/index.mvp/id/1796
https://datasheets.maximintegrated.com/en/ds/DS18B20.pdf
https://www.adafruit.com/datasheets/DS18B20.pdf

Thank you in advance.

alashkin, 2016-04-15:
This is a feature request, so could you give me more background on why do you need that? What is the project, and how that functionality will be used?

lukevenediger, 2016-05-10:
+1 - I have a need to integrate with a DS18B20 temperature sensor and this would make it a whole lot easier.

dvv, 2016-05-16:
+1 for the same reason. TIA

codelinq, 2017-03-15:
+1 for the same reason. Since there are c libraries written for onewire and ds18b20, would it be possible to include the OneWire.h file and write js wrapper using mjs's ffi to call the c functions? Or is that just wishful thinking? Here is the arduino OneWire library. https://github.com/PaulStoffregen/OneWire/blob/master/OneWire.cpp

meteofelix, 2017-03-17:
+1 also for being able to use the DS18B20

El-Mango, 2017-03-19:
+1 also for being able to use the DS18B20

cristiancs, 2017-04-26:
Any news on this?

cpq, 2017-04-26:
The PR is in the review, expect a commit tomorrow.

cpq, 2017-05-09:
https://github.com/cesanta/mongoose-os/tree/master/fw/examples/c_arduino_onewire

closed 2017-05-09 by cpq