cesanta/mongoose-os #118: continuous reboot

carterw, 2016-04-13:
I decided to try out the most recent commits so I went through the procedure for extracting and building the firmware, and was able to flash it with no problem.

The first thing I did after that was;
Wifi.setup("yourssid", "youpassword")

The device joined my network and all seemed well. Unfortunately, after a reset the device reboots over and over.

alashkin, 2016-04-14:
@carterw 

Hi, I cannot reproduce this, so, I need any details.
First of all, what esp model do you use? What size of its flash?

carterw, 2016-04-14:
According to the case markings it is a;
ESP-12
ESP8266MOD

alashkin, 2016-04-14:
I.e. it has 4MB flash?
Strange.
Ok, could you past here FNC output? Several reboots loop.
There is no crash dumps, right?

carterw, 2016-04-14:
There's no crash dump, it simply reboots over and over. I'll paste in the FNC output later today.

alashkin, 2016-04-14:
btw, you can try to reflash with `--esp8266-flash-size =4M`

carterw, 2016-04-14:
Will do.

alashkin, 2016-04-14:
It is FNC command line options

Run FNC --help (if you don't use Windows)

Also, you can go to Settings -> Advanced settings -> Check "esp8266-flash-size" and put 4M in this field.

carterw, 2016-04-15:
Flashing with the 4M option eliminated the problem, thanks. Device joined my wifi network after Wifi.setup() and not rebooting. Might want to include this tip in the documentation.

alashkin, 2016-04-15:
Good to hear that it helped.
I put this tip into FAQ
https://github.com/cesanta/smart.js/blob/master/docs/faq/intro.md#q-i-flashed-esp-module-with-smartjs-it-is-successfully-booted-and-joined-wifi-but-after-reset-the-device-reboots-over-and-over

closed 2016-04-15 by alashkin