cesanta/mongoose-os #125: Problem With Boot Loop ESP12E

prinzipx, 2016-05-01:
Hello,
i'm really new to this, so please excuse my poor english and my questions.
I have a ESP8266-12E chip and the power Supply is via 3,3 V Raspberry Pi.
After Installation and after the commands "`Wifi.setup('wlanname', 'wlanpw')`" and "`registerDevice()`", the ESP reboots every 10 Seconds.
I have attached some screenshots from the boot-process, is it a problem from the Power supply (because of the "` ets Jan  8 2013,rst cause:2, boot mode:(3,7)`")?
Can someone help me please? Thanks a lot.

Kind regards,
David

[image]

[image]

prinzipx, 2016-05-01:
Hello again,
i forgot that i was not able to flash the chip via the FnC so i flashed it via "`ESP FLASH DOWNLOAD TOOL`".
I dont know if this information is relevant.

[image]

prinzipx, 2016-05-01:
Hallo,

now (after breakfast ;-) ) the ESP have stopped rebooting. Here a new Screenshot.
[image]
I'm really sorry to post so much here, please close the issue because i think its a problem of me and not of mongoose iot :-)

alashkin, 2016-05-02:
Hi,
```
After Installation and after the commands "Wifi.setup('wlanname', 'wlanpw')" and "registerDevice()", the ESP reboots every 10 Seconds.
```

Looks like, that ESP was flashed with wrong flash-size parameters
(see here for details: https://github.com/cesanta/mongoose-iot/blob/master/docs/faq/intro.md#q-i-flashed-esp-module-with-mongoose-iot-it-is-successfully-booted-and-joined-wifi-but-after-reset-the-device-reboots-over-and-over)

But anyway, as I understand, FNC doesn't work for you.
You use Windows 10, right?

alashkin, 2016-05-13:
Hello,
How it is going with this issue?

mmikulicic, 2016-05-13:
we recently fixe an issue with saving config files that caused a crash loop also after saving wifi config.
it might be related

alashkin, 2016-05-13:
@prinzipx 
Ok, I'm closing this issue, feel free to reopen

closed 2016-05-13 by alashkin