cesanta/mongoose-os #128: reboot ( by wdt ) when https connect

vence2013, 2016-05-10:
//the code of app.js
var server = Http.createServer(function(req, res) {
  print('there is a http server!');
  //res.writeHead(200, {'Content-Type': 'text/plain'});
  res.write("hello world!");
  res.end('\n');  
}).listen("443", {ssl_cert:'cert.pem'});

the log file: https://gist.github.com/vence2013/de2007fd7df2092e2863d99e36c1dac8

vence2013, 2016-05-11:
while i use 'rsa:1024' cert.pem, it seems work well. previous pem is 'rsa:2048', esp8266 reboot while https connection, so, is there any suggestion parmeter like 'rsa' when generate a cert?

alashkin, 2016-05-13:
SSL handshake with RAS 2048 takes ~12 sec, while default WDT timeout was 10 sec.
I increased default timeout to 15 sec in https://github.com/cesanta/mongoose-iot/commit/e4d3695dd0aa6cd47bc8edcd581f5a75e3a3aae2 and small description of WDT dealing here: 
https://github.com/cesanta/mongoose-iot/blob/master/docs/faq/intro.md#q-my-esp-device-is-rebooted-by-watch-dog

closed 2016-05-13 by alashkin