cesanta/mongoose-os #129: ESP8266 AP mode - unstable (unusable)

IgorTrud, 2016-05-21:
AP mode on ESP8266 seems to to unstable ( and not usable).

I use this confing
```
{
  "wifi": {
    "sta": {
      "enable": false,
      "ssid":"SH_Host",
      "pass": "xxxxx"
    },
    "ap": {
      "enable": true,
      "keep_enabled": true,
      "trigger_on_gpio": -1,
      "ssid": "SH_LocalHost",
      "pass": "xxxx",
      "hidden": false,
..........
```

Then, I seach wifi "SH_LocalHost" and try to connect from my phone.
Only 2 of 10 connections successfull.  Most trys hang during  "Getting IP adress" or fails with "wrong password". 
Sometimes, network  is  not visible in scan for some time.
Module is not rebooting during this trys.
If wifi connection happens, there is still a problem: connection live 2-3 minuts, than die.
Phone start to reconnect to wifi, and fails.

I check this on sevral ESP-03 modules, and on ESP-01 - all the same.
Also, modules powerd from diffrent power source, including a very good one,
so  it does not look like a HW-related problem. 

Any comments or thoughts ?

IgorTrud, 2016-05-21:
Foget to say: Client mode  work fine on the same HW.

alashkin, 2016-05-26:
Hello,
I just tried and everything works fine for me (i know, it means nothing)
But, at first, could you try the latest sources?
I fixed problem with AP in https://github.com/cesanta/mongoose-iot/commit/3a06147aa27c8f8150093acc850f493d9e8e517b
Not sure it is related, but still.

IgorTrud, 2016-05-30:
No, just the same - unstable.
Going to try on ESP-12 module, waiting for my order.

alashkin, 2016-06-01:
Since I'm not able to reproduce this issue,..  Let's try to think.
1. What flash size your devices have? 
2. Do you run some specific JS code during your connections attempts?
3. Could you share debug output here? Please, make ensure, what debug config looks like this:
```
  "debug": {
    "level": 3,
    "stderr_uart": 0
  }
```
so, level is 3 and errors redirected to UART0

cpq, 2017-02-07:
@IgorTrud could you try now please?

cpq, 2017-02-09:
closing as rotten,
please reopen if needed.

closed 2017-02-09 by cpq