cesanta/mongoose-os #130: GPIO.setISR handler pulse duration

dvv, 2016-05-23:
Hello!

Say I want to implement a protocol decoder (e.g. IR NEC, DHT*, 1-Wire etc.) in an ISR-driven way.
What is the minimum pulse duration that `GPIO.setISR` can report?

TIA,
--Vladimir

alashkin, 2016-05-26:
Tbh, it is hard to say.
How it works: FW _disables_ interrupts, calls callback provided in setISR and enables interruptions back only after callback is completed.
Thus, JS timing is more-more-more important comparing to ability of ESP8266.
So, the only way to get numbers, is to make an experiment with _your_ specific code in callback.

mmikulicic, 2016-05-26:
I think low level protocol decoders should be written in C and offer an JS API, for example like we do for I2C and SPI

dvv, 2016-05-26:
@alashkin Right. Am wondering what is the numbers in case on _empty_ callback. (Have no ESP within the grasp, am thinking abstractly so far).

@mmikulicic Probably right. But I'd like to try to apply the goods provided by coding in interpreting language.

dvv, 2016-05-28:
A rough estimation of `Sys.time` granularity:

```js
for (var i = 0; i < 10; ++i ) console.log(Sys.time() - Sys.time())

-0.000223
-0.0002099999
-0.000207
-0.0002109999
-0.0002069999
-0.000211
-0.0002059999
-0.0002139999
-0.000202
-0.000207
```

NB: a slight speedup via caching seems to work:

```js
var fn = Sys.time
for (var i = 0; i < 10; ++i ) console.log(fn() - fn())

-0.0001829999
-0.000159
-0.000155
-0.0001589999
-0.000156
-0.000159
-0.0001549999
-0.0001629999
-0.000152
-0.000155
```

So it seems we've no JS means to distinguish signals faster than 200us.
But it's pretty good for decoding say IR NEC ;)

dvv, 2016-05-28:
Sampling with memorizing seems to be much worse:

```js
var fn = Sys.time
var ts = [0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
for (var i = 0; i < 10; ++i ) ts[i] = fn()
for (var i = 1; i < 10; ++i ) console.log(ts[i] - ts[i - 1])

0.000774
0.000938
0.0005659999
0.000683
0.0005669999
0.000606
0.000602
0.000609
0.0005659999
```

mmikulicic, 2016-05-28:
Try to build it with:

1. STACK_GUARD=false
2. GC_DEBUG=false

These checks are enabled by default but are quite expensive.

Also could you please compare `print` with `console.log`. (`console.log` has some overhead because it also allows you send logs to a server and queue logs on the flash when the connection is unavailable).

Another source of slowness is the current flash memory mapping.
How are you uploading that snippet of code ?
Are you entering it from the prompt ?

> Sampling with memorizing seems to be much worse:
yeah, currently the array grows one slot at a time, and arrays are currently implemented in a very simple and unoptimized way. For every assign a new property will be allocated from the JS heap; furthermore array lookups are slow.

We have a few optimizations on our roadmap, but for now we're focusing on correctness

dvv, 2016-05-28:
You see, here's no issue with `console.log` vs. `print` as I measure the time taken by just two calls to `Sys.time` and one subtraction. Futhermore, to cope with possibly slower dereferencing I cache `Sys.time` function.

What to arrays as you can see I preallocated the array to be sure no allocations should occur during the run.

So this issue transformed into kinda research for reference.

What I would very like to have is a function implemented in C which asyncronously reads a number of pin state transitions and callback with the array of pulse durations, kinda:
```js
GPIO.sample(pin, ntransitions, timeout, function (err, durations) { /* we have the whole JS to analyse durations here */ });
```

closed 2017-02-09 by cpq