cesanta/mongoose-os #132: console.log.* files in filesystem

dvv, 2016-05-27:
Wonder what these console.log.* files are?

```
$ File.list('.')
["conf_app_defaults.json","conf_app_schema.json","index.html","conf_sys_defaults.json","I2C.js","sys_init.js","conf_sys_schema.json","conflib.js","demo.js","ca.pem","console.log.0000000001","console.log.0000000002","console.log.0000000003","console.log.0000000004","console.log.0000000005","console.log.0000000006","console.log.0000000007","console.log.0000000008","console.log.0000000009","console.log.0000000010","console.log.0000000011","console.log.0000000012","console.log.0000000013","console.log.0000000014","console.log.0000000015","console.log.0000000016","console.log.0000000017","console.log.0000000018","console.log.0000000019","console.log.0000000020","console.log.0000000021","console.log.0000000022","console.log.0000000023","console.log.0000000024","console.log.0000000025","console.log.0000000026","conf.json"]
```

I've run a test script which used `console.log` massively which apparently led to a lot of garbage in console, and cyclic reboot with message `Failed to load configuration from: conf.json Error: {"message":"cannot open [conf.json]",...`

Firmware: mongoose-iot-20160426093822

So we somehow use flash as `console.log` buffer?
If yes how to turn it off?

TIA

dvv, 2016-05-27:
**UPDATE**

This should disable log caching:
```
Sys.conf.console.send_to_cloud = false
Sys.conf.save()
```
Is there any reason why `send_to_cloud` is not configurable via configuration page?

dvv, 2016-05-27:
Below is the script I used:
```js
var sub = dgram.createSocket("udp4");
sub.on("message", function (message, peer) { console.log("receive", GC.stat().sysfree, message.length); });
sub.bind(5684, "0.0.0.0");
```
Loaded with nodejs:
```js
function sss(len) {
  var s = "[";
  for (; --len >= 0;) s += ".";
  s += "]";
  return s;
}

var fd = require("dgram").createSocket("udp4");
function out() {
  var buf = new Buffer("foo" + "=" + sss(1000)); // load pedal
  fd.send(buf, 0, buf.length, 5684, "192.168.1.201", function (err, bytes) {
    if (err) throw err;
    console.log("SENT");
    setTimeout(out, 20) // accelerator pedal
  });
}
out();
```

alashkin, 2016-06-01:
>Wonder what these console.log.* files are?

`console.log` prints output to serial but also sends it to cloud, thus, logs can be retrieved by clubby command. If there are too many unsent logs to fit in memory, `console.log` puts it on flash to send later.

>Is there any reason why send_to_cloud is not configurable via configuration page?

No, it is bug, fixed in a90165486758752f76a766c60c92a1c3797bedb3

closed 2016-06-01 by alashkin