cesanta/mongoose-os #135: js interpreter causes a crash if last line is a comment

carterw, 2016-06-26:
This cost me an hour to figure out WTF was going on. I am using FW I built and flashed today. Put a comment as the last line of foo.js, ship it to the 8266, and then File.eval('foo,js');

function whatever() {
    var a = 1;
}

// if last line is a comment the 8266 crashes

----------------- on the console it says this just prior to core dump
MMAP invalid address 0xbc30bce: block 1, off 0, pages 1, desc 9

Trap 9: pc=0x40101d5a va=0x1

dimonomid, 2016-06-26:
Yes we're aware of the issue with crashing on MMAP invalid address, hovewer, the cause is not the comment on the last line, but certain file sizes. So, the "workaround" is to add some more comment, to force file size to change.

We'll fix it in several days.

And, btw, the example you provided
```
function whatever() {
var a = 1;
}

// if last line is a comment the 8266 crashes
```
Doesn't really cause my ESP8266 to crash. Could you please confirm that exactly this file causes it to crash for you?

-----

Side note: you can save RAM if you put your custom JS files not in `fw/src/fs`, but in `fw/src/js`, and build it with the command like: `make -C fw/platforms/esp8266 JSBIN_EXTRA="foo.js bar.js"`: then, files `foo.js` and `bar.js` will be precompiled at your host, and the firmware will contain bytecode in flash memory instead of JS code. So, when you execute it, it's not needed to compile and allocate RAM to store bytecode.

carterw, 2016-06-26:
Wow that's a good tip about embedding programs, I will try it.

My experience here is that the program above does cause a crash. The file size doesn't matter, larger or smaller files also crash.

dimonomid, 2016-06-26:
> My experience here is that the program above does cause a crash. The file size doesn't matter, larger or smaller files also crash.

Hm, I'm confused. I can't reproduce the issue: I have a file `foo.js` with the contents above, build&flash the firmware, and after `File.eval('foo.js')`, I have the function `whatever` defined; no crash.

Anyway, after the bug with file sizes is fixed, we'll ask you to check if this issues is resolved or not.

dimonomid, 2016-06-28:
@carterw , we fixed the bug which caused fw to crash on certain file sizes. Could you please pull latest master, make clean build and check if this issue with last-line-comments is still present.

If yes, then I'd like to hear any changes you've made to our firmware: any configuration changes, any custom JS files added, etc; since I can't reproduce it on my build.

carterw, 2016-06-28:
Thanks, I will try it later today and provide that feedback.

carterw, 2016-06-29:
I made fresh FW and flashed it. No problem with last-line-comments now so apparently your fix solved that problem as well, thanks.

closed 2016-06-29 by carterw