cesanta/mongoose-os #140: Error deploying firmware: context deadline exceeded

kroshilin, 2016-07-10:
On ESP8266
Every time I push flash button in the IDE, i recieve this messages
...
[ 09:00:51 ] == Build success
[ 09:00:51 ] == Deploying firmware
[ 09:01:23 ] == Error deploying firmware: context deadline exceeded

So, I could not save any file to EPS8266
There is screenshot with IDE - https://yadi.sk/i/TSBQfXLUt9tZu

cpq, 2016-07-11:
Привет Антон,
this error is not fatal. Give some time. Do you see a hello message from a device in the log window?

kroshilin, 2016-07-11:
Привет!
No, i don't. Because it is only the initial FW version is on my device. I suppose it does not have any files that can console.log("Hello")?

cpq, 2016-07-11:
Once you clicked "Run" button in the IDE, a new firmware should be saved on your device (only things that have changed are updated). That new firmware has a "Hello from device control tutorial!" log command, which should be sent to the cloud and displayed in your browser.

Can you send all log messages that are shown in your browser please?

kroshilin, 2016-07-11:
I think that the problem is that new FW with changes was not deployed to my device. 
[ 09:01:23 ] == Error deploying firmware: context deadline exceeded

Build log
`[ 18:15:29 ] make: Entering directory `/code'

[ 18:15:29 ] /sjs/repo/fw/platforms/esp8266/Makefile.build:441: warning: overriding commands for target `/sjs/repo/fw/platforms/esp8266/.build/fw_temp/0x00000.bin'

[ 18:15:29 ] /sjs/repo/fw/platforms/esp8266/Makefile.build:292: warning: ignoring old commands for target `/sjs/repo/fw/platforms/esp8266/.build/fw_temp/0x00000.bin'

[ 18:15:29 ] make[1]: Entering directory `/sjs/repo/common/platforms/esp8266/lwip'

[ 18:15:29 ] make[1]: Nothing to be done for `all'. make[1]: Leaving directory `/sjs/repo/common/platforms/esp8266/lwip'

[ 18:15:29 ] # OSX virtualbox shared folder have issues with some symlinks. JS_SRCS contains their canonical location

[ 18:15:29 ] V7C /sjs/repo/fw/src/js/sys_init.js -> /sjs/repo/fw/platforms/esp8266/.build/fs_temp/sys_init.js

[ 18:15:29 ] V7C /sjs/repo/fw/src/js/demo.js -> /sjs/repo/fw/platforms/esp8266/.build/fs_temp/demo.js

[ 18:15:29 ] V7C /sjs/repo/fw/src/js/I2C.js -> /sjs/repo/fw/platforms/esp8266/.build/fs_temp/I2C.js

[ 18:15:29 ] V7C /sjs/repo/fw/src/fs/conflib.js -> /sjs/repo/fw/platforms/esp8266/.build/fs_temp/conflib.js

[ 18:15:29 ] V7C /code/filesystem/app.js -> /sjs/repo/fw/platforms/esp8266/.build/fs_temp/app.js

[ 18:15:29 ] RSYNC -> /sjs/repo/fw/platforms/esp8266/.build/fs_temp/

[ 18:15:29 ] sending incremental file list

[ 18:15:29 ] app.js

[ 18:15:29 ] ca.pem

[ 18:15:29 ] conf_app_defaults.json

[ 18:15:29 ] conf_app_schema.json

[ 18:15:29 ] conf_sys_defaults.json

[ 18:15:29 ] conf_sys_schema.json

[ 18:15:29 ] index.html

[ 18:15:29 ]

[ 18:15:29 ] sent 40,613 bytes received 149 bytes 81,524.00 bytes/sec total size is 40,174 speedup is 0.99

[ 18:15:29 ] FS /sjs/repo/fw/platforms/esp8266/.build/fs_temp (131072 @ 0xdc000)-> /sjs/repo/fw/platforms/esp8266/.build/fw_temp/0xdc000.bin

[ 18:15:29 ] adding files in directory /sjs/repo/fw/platforms/esp8266/.build/fs_temp

[ 18:15:29 ] a conflib.js

[ 18:15:29 ] a conf_sys_schema.json

[ 18:15:29 ] a ca.pem

[ 18:15:29 ] a conf_sys_defaults.json

[ 18:15:29 ] a index.html

[ 18:15:29 ] a I2C.js

[ 18:15:29 ] a conf_app_schema.json

[ 18:15:30 ] a demo.js a conf_app_defaults.json a app.js

[ 18:15:30 ] a sys_init.js

[ 18:15:30 ] Image stats: size=131072, space: total=113201, used=54969, free=58232

[ 18:15:30 ] GEN /sjs/repo/fw/platforms/esp8266/.build/fw_temp/manifest.json

[ 18:15:30 ] ZIP /out/mongoose-iot-esp8266-last.zip

[ 18:15:30 ] Add /sjs/repo/fw/platforms/esp8266/.build/fw_temp/0x01000.bin

[ 18:15:30 ] Add /sjs/repo/fw/platforms/esp8266/.build/fw_temp/0x00000.bin

[ 18:15:30 ] Add /sjs/repo/fw/platforms/esp8266/.build/fw_temp/0xdc000.bin

[ 18:15:30 ] Add /sjs/repo/fw/platforms/esp8266/.build/fw_temp/0x11000.bin

[ 18:15:30 ] Built version 20160711075518

[ 18:15:30 ] make: Leaving directory `/code'

[ 18:15:32 ] == Build success

[ 18:15:32 ] == Deploying firmware

[ 18:16:07 ] == Error deploying firmware: context deadline exceeded `

Device console
`[ 18:15:32 ] Starting update

[ 18:15:32 ] Updating FW `

cpq, 2016-07-11:
Apparently updater failed - @alashkin what do you think is the problem?

alashkin, 2016-07-11:
Hi,
I tried several time and once I've seen a picture, what the device was unable to connect WiFi.
Could you reset it a couple of time, for the beginning?

And usual support question: what ESP model, what size of flash, did you try to flash it via esptool/mft/any other serial flasher?

kroshilin, 2016-07-11:
I've reset it and left switched on. 
My ESP is like [this one](http://ru.aliexpress.com/item/2pcs-lot-ESP8266-Serial-Port-WIFI-Wireless-Transceiver-Send-Receive-Module-IO-Lead-Out-ESP-201/32267630006.html?spm=2114.13010608.0.119.1JaaNB)

alashkin, 2016-07-11:
ESP-201? 
What is actual flash size? Can it be successfully flashed with MFT?

kroshilin, 2016-07-11:
Dont know flash size. How could I refine it? Yes, i flashed it with MFT successfully.

cpq, 2016-12-27:
Closing as obsolete.

closed 2016-12-27 by cpq