cesanta/mongoose-os #142: mongoose IDE: Links don't work in API page

wallaroo, 2016-07-18:
in the api page (https://mongoose-iot.com/profile/api/) of the Mongoose IDE links to resources don't works.
Tested in chrome and firefox.

error in console:
Error: Could not resolve 'main.project.api' from state

mmikulicic, 2016-07-18:
Could you please try https://mongoose-iot.com/profile/api// ?

How did you get the https://mongoose-iot.com/profile/api/ link ? The nav bar points to the former.

wallaroo, 2016-07-18:
hi!
same result with https://mongoose-iot.com/profile/api//:
the page remains on /v1/Account api detail.
and for each click on other api the following error is thrown:
`vendor-c16bc78355.js:32 Error: Could not resolve 'main.project.api' from state 'main.profile.api'
    at Object.C.transitionTo (vendor-c16bc78355.js:57)
    at Object.C.go (vendor-c16bc78355.js:57)
    at vendor-c16bc78355.js:57
    at vendor-c16bc78355.js:33
    at o (vendor-c16bc78355.js:31)
    at vendor-c16bc78355.js:31`

PS.
maybe i did a bad copy-paste for the missing '/'

mmikulicic, 2016-07-18:
yes the direct link is broken. 
can you access the API page if you click directly on the API icon on the left navigation bar ?

[image]

wallaroo, 2016-07-18:
this is the test case:
1) go to https://mongoose-iot.com/
2) signin with github
3)click on api button at the left navbar (that of your image)
4) click on "CreateUser" (or whatever other api link)
result:
the page remain on initial /v1/Account api detail and the following javascript error is thrown:
`vendor-c16bc78355.js:32 Error: Could not resolve 'main.project.api' from state 'main.profile.api' at Object.C.transitionTo (vendor-c16bc78355.js:57) at Object.C.go (vendor-c16bc78355.js:57) at vendor-c16bc78355.js:57 at vendor-c16bc78355.js:33 at o (vendor-c16bc78355.js:31) at vendor-c16bc78355.js:31`

mmikulicic, 2016-07-18:
ah, sorry; now it's clear; fixing

wallaroo, 2016-08-10:
now all works well

closed 2016-08-10 by wallaroo