cesanta/mongoose-os #145: Exec error: "Wifi.mode is not a function"

carterw, 2016-07-28:
In the API reference I am seeing;
Wifi.mode(mode) -> true or false: Set Wifi mode. mode is a number, 1 is station, 2 is soft-AP, 3 is station + soft-AP

But typing in to the console this I get an error;
Wifi.mode(1)
Exec error: "Wifi.mode is not a function"

Is this API function no longer necessary?

mmikulicic, 2016-07-29:
We converted that API to a configuration entry:

`Sys.conf.wifi.ap.enable` and `Sys.conf.wifi.station.enable`

You can configure it either manually (and save&reboot with: `Sys.conf.save()`) or via the configuration web server by pointing your browser to the device.

closed 2016-07-29 by mmikulicic

carterw, 2016-07-29:
You might want to update your API doc. Would be great if you would document everything that gets adjusted in Sys.conf and what the default behavior is after a fresh reflash...

mmikulicic, 2016-07-29:
yeah, the documentation has plenty of room for improvement.

For now, this feature is mentioned in:

1. mgiot config page is self-documenting. Each option is described in https://github.com/cesanta/mongoose-iot/blob/master/fw/src/fs/conf_sys_schema.json and the documentation rendered along the control
2. We mention this config setting is mentioned in the bootstrap documentation at: https://docs.cesanta.com/mongoose-iot/master/#/firmware/boot.md/
3. The Wifi API (at https://docs.cesanta.com/mongoose-iot/master/#/javascript-api/wifi.md/) doesn't mention that because there is no API to set it
4. Config C API: https://docs.cesanta.com/mongoose-iot/master/#/c-api/config.md/
5. Config JS API: missing ! :-( mentioned briefly in https://docs.cesanta.com/mongoose-iot/master/#/firmware/filesystem.md/