cesanta/mongoose-os #146: JavaScript precompile problem

carterw, 2016-07-31:
I am running out of RAM now that initial free RAM has dropped to 25720 and I have to allocate at least 8K to_V7_STRING_BUF_RESERVE in order to exchange even small messages.

A few weeks ago this tip was posted;
"Side note: you can save RAM if you put your custom JS files not in fw/src/fs, but in fw/src/js, and build firmware with the command like: make -C fw/platforms/esp8266 JSBIN_EXTRA="foo.js bar.js": then, files foo.js and bar.js will be precompiled at your host, and the firmware will contain bytecode in flash memory instead of JS code."

I am trying to have at least one program file be precompiled but the build fails. When I copy foo.js into fw/src/fs and then;
make -C fw/platforms/esp8266 JSBIN_EXTRA="foo.js"

I get the message "*** No rule to make target 'foo.js', needed by " ...

carterw, 2016-08-02:
No help from dev team on this? It seems pretty fundamental.

Looking at the makefile (had to dig it out, .../fw/platforms/esp8266/Makefile.build) it turns out that what you actually have to invoke is ;
make -C fw/platforms/esp8266 JSBIN_EXTRA="full-pathname-to-your-file"

All this really does, as far as I can tell, is cause your .js file to be precompiled and put into the filesystem when the device is flashed. You still have to eval it in app.js or somewhere, and I am not noticing that it saves any memory.

mmikulicic, 2016-08-02:
Hi, sorry we're somewhat busy on other fronts.

w.r.t the memory saving:

1. precompiling allows running code even if you build the VM with `V7_NO_COMPILER`, especially useful on small embedded platforms where the code has to be loaded on ram (e.g. cc3200)

2. evaluating a precompiled JS file on esp8266 will mmap the file straight from flash. We have a simple mmap shim in software that emulates all memory reads in a given memory area and remaps them to mmapped flash memory reads in the SPIFFS filesystem (which divides the files in non contiguous blocks)

When the code is evaluated some parts of it have to be loaded in memory (furthermore the mmapping layer requires a few bytes of overhead), so you might not notice a big difference in usage if your precompiled file is very small.

As for the JSBIN_EXTRA: our makefiles could be improved a lot....
the idea was that you put the js files inside your `fw/src/fs` dir, but I guess full pathnames should work as well.

We're working on a cmdline util that will help you build a firmware with less contortions.

carterw, 2016-08-02:
Thanks for the info! The file I had tried was in fact small. I'll try something larger and compare.

mmikulicic, 2016-08-02:
Please let us know, if it doesn't show it's a regression and I'd really like to know about it :-)

carterw, 2016-08-02:
Speaking of regressions, do you guys have a regression test that you can run against your firmware?  Something that exercises all of the API and also the JavaScript interpreter? If not I highly recommend it. It would be a valuable resource for developers like me as well.

mmikulicic, 2016-08-02:
v7 tests are in https://github.com/cesanta/v7/tree/master/tests
we have a bunch of v7 specific tests and we also run it against the official ECMA5 test suite.

as for the mongoose-iot tests ... it's complicated: 

Unit tests would be hard given that the code depends on the device SDKs and the produced output has to run on those embedded devices. So what we have is an integration test suite, basically a set of end-to-end test that we run in our continuous integration system that is composed of a few embedded devices connected to a rpi3 host. We test networking (client and server), OTA, basic GPIO functionality. The test code on the device actually talks to other devices via the network and actually flips some GPIOs that the rpi host checks. We even have some SPI devices connected to the device under test. 

The setup is highly customized and depends on a quite few things outside of this repostory and so we decided to not invest a lot of time to expose it and document it given that it would be hard for an external user to reproduce the same testing rig anyway.

There are a few things that we test only indirectly: for example if suddenly the mmapping code gets  butchered and precompilation doesn't work again, all of our device tests will fail given that the bulk of the testing code is written in JS and thus will OOM the device before the test could even start.

That said, we could get very unlucky and break precompilation in just the right way that it doesn't show up in our tests :-) That's why I'm curious about your comment :-)

carterw, 2016-08-02:
Definitely saves memory for larger files. I am not sure exactly how much, maybe 50%. That helps a lot.

mmikulicic, 2016-08-03:
could you please share the JS file you used as a test ?

cpq, 2016-12-28:
Closing this as obsolete.
Note we're working on JS engine refactoring which is expected to be in working condition by the end of Jan.
Main difference with V7: smaller footprint, way less complex, many standard JS features are NOT supported, ability to load C functions directly without writing glue code.
See https://github.com/cesanta/mjs
Initial support is already there

closed 2016-12-28 by cpq