cesanta/mongoose-os #147: GPIO.setISR() allows only one invocation per pin?

carterw, 2016-08-06:
I'm experimenting with GPIO.setISR() and finding that I can call it one time to set a transition mode and a function callback for a pin. But if I try to call it again on that pin I get an error;
"Only one interruption handler is allowed for pin"

Is this by design or am I doing something wrong? It seems like I should be able to turn it off or set it to something different.

GPIO.setISR(0, GPIO.ANYEDGE, function() {
			print('pin ' + pin + ' transitioned');
		});  // succeeds

GPIO.setISR(0, GPIO.OFF, function() {});  // fails!

carterw, 2016-08-06:
Another odd thing. If I set;
GPIO.setISR(0, GPIO.LOW, function() {
print('pin ' + pin + ' transitioned');
}); // succeeds

and then later...
GPIO.write(0, false);

cpq, 2017-02-07:
The GPIO API has changed:
https://github.com/cesanta/mongoose-os/blob/master/fw/examples/mjs_hello/fs/api_gpio.js

Now, it is an FFI-ed C API.
See https://mongoose-iot.com/blog/mjs-a-new-approach-to-embedded-scripting/ for more info and rationale of FFI.

Please reopen if needed.

closed 2017-02-07 by cpq