cesanta/mongoose-os #148: dsleep() never wakes up?

carterw, 2016-08-06:
Invoking this on the console;
dsleep(1000)

... causes pin 0 to go low which is fine. Presumably the device would wake up after a second but it never wakes up. Unless the reset button is pressed or power cycled.

carterw, 2016-08-06:
Update - this experiment appears to have damaged my development board. Invoking Wifi.setup() from the console now results in a number of connect attempts and disconnects (which I had never previously seen) and eventually a crash. This happens even with release 1.1 firmware.

This is Mongoose IoT Firmware 1.1 (esp8266)
......
f 0, scandone
no ATT6y9a4r6 found, reconnect after 1s
sj_wifi_on_change_cb Wifi: disconnected
reconnect
f -180, E:M 32 (5 blocks)

Trap 29: pc=0x4029f41e va=0x0
Dumping core
--- BEGIN CORE DUMP ---
{"arch": "ESP8266", "cause": 29, "REGS": {"addr": 1073653352, "data": "YushQCD+

carterw, 2016-08-09:
usleep() doesn't appear to do anything. Typing in usleep(200000) on the MFT console and I still get prompts when I hit return. And my running software continues to run.

Am I missing something? Has nobody else tried these things? Do I need to tie some pins to power or ground to make it work?

wallaroo, 2016-08-10:
Usleep takes microseconds as parameter... So 1 second = 1000000

mmikulicic, 2016-08-10:
yeah. the `u` in `usleep` stands for micro, from the greek letter μ (Mu)

The API is modeled after a pretty standard UNIX function of the same name: http://linux.die.net/man/3/usleep

mmikulicic, 2016-08-10:
> Update - this experiment appears to have damaged my development board. 
@carterw do you have any updates on that; does you board still work ?

carterw, 2016-08-10:
Powercycling the board brought things back to normal. I am guessing that the radio was turned off somehow.

But dsleep() is still a mystery. Does it work? If so please explain how to use it.

mmikulicic, 2016-08-10:
IIRC, deep sleep basically powers down the board while keeping the RTC on and configures the RTC to signal a pin when some amount of time passes. That pin should be connected to the device reset line. You should do the wiring yourself, otherwise dsleep basically just halts the device forever.

@carterw does device just do nothing when you invoke `dsleep()`

@alashkin can you please fill in the details? Can you please check whether dsleep (at least the C api) works as intended ?

realzeus, 2016-08-10:
@carterw As far as I know dsleep requires a special wiring (at least on some boards).
Please take a look here: [http://tim.jagenberg.info/2015/01/18/low-power-esp8266/](http://tim.jagenberg.info/2015/01/18/low-power-esp8266/)

wallaroo, 2016-08-10:
i tested dsleep with the 1.0-rc1 and esp8266 and all worked well...
you have to wire GPIO16 with RST otherwise the board can't wake up.
NB. also dsleep use microseconds

carterw, 2016-08-10:
Thanks for the tip, I'll try it with that wiring. Does anyone know what the available "dsleep_option" values are and what they do?

wallaroo, 2016-08-10:
take a look to http://www.espressif.com/sites/default/files/9b-esp8266-low_power_solutions_en_0.pdf (page 6)
that option define if and how RF is powered on next wake up

mmikulicic, 2016-08-10:
http://russ.russmathis.com/esp8266-power-modes/

```
     RF_DEFAULT = 0,  //RF_CAL depends on init data from byte 108
     RF_CAL = 1,  //RF_CAL enabled causes large current drain after wake 170mA
     RF_NO_CAL = 2,  //RF_CAL disabled, small current drain after wake 75mA
     RF_DISABLED = 4  //RF disabled, smallest current drain after wake 15mA
```

You might also want to consult the official doc from espressif at:

http://www.espressif.com/sites/default/files/2a-esp8266-sdk_getting_started_guide_en_0.pdf

look for the `system_deep_sleep_set_option` API function documentation.

BTW, we're starting to use ESP866 SDK 2.0 from today in master.

carterw, 2016-08-14:
I cloned and flashed firmware today, presumably using ESP866 SDK 2.0, and all seems to be working as expected.

I wired pin 16 to reset, and dsleep(10000000, 1) does cause the 8266 to wake up after 10 seconds. It appears to wake up as if it were reset, which is to be expected I guess.

On the other hand, usleep(10000000) causes a core dump on wakeup;

system_restart_local WDT reset, info: exccause=4 epc1=0x40002ef1 epc2=0x00000000 epc3=0x00000000 vaddr=0x4029d035 depc=0x00000000
Dumping core
--- BEGIN CORE DUMP ---
{"arch": "ESP8266", "cause": 100, "REGS": {"addr": 1073651444, "data": "8S4AQLD6/

mmikulicic, 2016-08-14:
the watch dog timer kicked in.

carterw, 2016-08-14:
Ah, so its on by default. I called Sys.wdtSetTimeout(timeout) with a timeout longer than the sleep time and got no core dump on wakeup.

cpq, 2016-12-27:
Closing this.

closed 2016-12-27 by cpq