cesanta/mongoose-os #149: fw make error

wallaroo, 2016-08-07:
hi all, i'm trying to build mongoose-iot but i get this error:
```
make -C platforms/posix
make[1]: Entering directory '/root/mongoose/fw/platforms/posix'
make[1]: Warning: File 'bin' has modification time 69832 s in the future
  CC    main.c -> .build/main.o
  CC    ../../src/sj_v7_ext.c -> .build/sj_v7_ext.o
  CC    ../../src/sj_init.c -> .build/sj_init.o
  CC    ../../src/sj_init_js.c -> .build/sj_init_js.o
  CC    ../../src/sj_ws_client_js.c -> .build/sj_ws_client_js.o
  CC    ../../src/sj_mqtt_js.c -> .build/sj_mqtt_js.o
  CC    ../../src/sj_gpio_js.c -> .build/sj_gpio_js.o
  CC    ../../src/sj_adc_js.c -> .build/sj_adc_js.o
In file included from ../../src/sj_adc_js.c:6:0:
../../src/sj_adc.h:11:1: error: unknown type name 'uint32_t'
 uint32_t sj_adc_read(int pin);
 ^
Makefile:174: recipe for target '.build/sj_adc_js.o' failed
make[1]: *** [.build/sj_adc_js.o] Error 1
make[1]: Leaving directory '/root/mongoose/fw/platforms/posix'
Makefile:9: recipe for target 'all' failed
make: *** [all] Error 2
```

repo cloned today (07/08/2016) on ubuntu 16.04

mmikulicic, 2016-08-08:
I tried to reproduce the issue with a simple and quick way to get a clean install of ubuntu 16.04:

```sh
$ docker run -ti --rm ubuntu:16.04
# apt-get update && apt-get install --yes build-essential git python python-git
# cd /tmp; git clone https://github.com/cesanta/mongoose-iot
# cd mongoose-iot
# make -C fw/platforms/posix
make: Entering directory '/tmp/mongoose-iot/fw/platforms/posix'
  MKDIR bin
  MKDIR .build
  GEN   .build/sys_config.c
  GEN   .build/sys_ro_vars.c
.....
  CC    ../../src/sj_adc_js.c -> .build/sj_adc_js.o
....
  CC    .build/build_info.c -> .build/build_info.o
LD bin/mongoose-iot
MKFS fs
make: Leaving directory '/tmp/mongoose-iot/fw/platforms/posix'
```

Somehow the right headers defining the standard `uint32_t` aren't being loaded. 
It doesn't look like a normal ubuntu 16.04 build. Are you using a cross-toolchain (do you have something in your `CC` env var) ? Is it possible that your headers might be corrupted ?

wallaroo, 2016-08-08:
i think i had some issue related to the mounted volume in my docker container (maybe something like EOL DOS) i've recloned in a different directory and all worked well...
thank you for support!!

closed 2016-08-08 by wallaroo