cesanta/mongoose-os #150: ADC values

carterw, 2016-08-08:
I am looking at the ADC API for the 8266, What do the numbers mean? How to calibrate? The pin here is attached to ground. Is ADC.read(6) returning millivolts?

ADC.readVoltage(6)
0.005859375

ADC.read(6)
5

The API doc says "Pin parameter is ignored on this platform" but you have to supply a 6.

mmikulicic, 2016-08-08:
hmm, not sure how the pin can make a difference.

```c
uint32_t sj_adc_read(int pin) {
  (void) pin;
  return 0xFFFF & system_adc_read();
}

double sj_adc_read_voltage(int pin) {
  return sj_adc_read(pin) / 1024.0; /* ESP8266 has a 10-bit ADC */
}
```

that's all what we do.

Please refer to the esp8266 and your actual board specs to see how this reading translates to the real world; IIRC some boards (nodemcu ?) provide a voltage divider resistor.

carterw, 2016-08-08:
Thanks for that info!

closed 2016-08-08 by carterw