cesanta/mongoose-os #151: Help: Flash Layout / FS Size

wallaroo, 2016-08-10:
hi, i'm trying to increase the filesystem size... i have an esp8266 with 4M of flash, so i would like to increase fs size from 128K to 1M (or better 3M).
i first tried to use advanced settings in the MFT and after i tried to change 
FLASH_SIZE ?= 4M
FS_SIZE ?= 1M
in Makerfile.build and rebuild but after flash Sys.fs.free() returns always ~60k
i think there is some calculation to do in order to set the correct offsets of addresses for the four images, but i can't figure out what is the current flash layout and what are the constraints... (i'm a web developer... i'm feeling behind the enemy lines :-))

someone can help me?

cpq, 2016-08-10:
@alashkin could you comment on this please?

mmikulicic, 2016-08-10:
The easiest way to see the current flash layout is just to peek inside the manifest.json file inside a firmware zip file.

The relevant bits:

```json
  "parts": {
    "boot": {
      "addr": 0,
      "cs_sha1": "d58f783adb68238ac682948811fa2a160638ae44",
      "size": 2512,
      "src": "0x00000.bin"
    },
    "boot_cfg": {
      "addr": 4096,
      "cs_sha1": "e0c66649d1434eca3435033a32634cb90cef0f31",
      "size": 4096,
      "src": "0x01000.bin"
    },
    "fs": {
      "addr": 901120,
      "cs_sha1": "a1c7b734f1e5ad7f2b91e20c611f1991ea951a82",
      "fs_block_size": 4096,
      "fs_erase_size": 4096,
      "fs_page_size": 256,
      "fs_size": 131072,
      "size": 131072,
      "src": "0xdc000.bin",
      "type": "fs"
    },
    "fw": {
      "addr": 69632,
      "cs_sha1": "a6f5cffdb9de67b2de4d5cfcab01d9716684cdba",
      "size": 624928,
      "src": "0x11000.bin"
    }
```

The address of the filesystem is computed as:

```
FS_ADDR  = $(shell printf "0x%05x" "$$((0x100000 - $(_FS_SIZE) - $(SYS_PARAM_SIZE)))")
```

Basically the FS is put towards the end of the flash. This way when the code section grows, the FS doesn't need to be relocated and our flashing tool has less things to reflash (it can avoid reflashing if the content doesn't change, by checksumming).

As you can see we hardcoded 0x100000 instead of FLASH_SIZE.

The reason for that is that our builds support Over The Air Updates (OTA).
You can perform unattended over the air updates to devices that are deployed out of your reach, and used by customers that don't have flashing tools installed; if an updates goes wrong, the device is effectively bricked as far as the end user is concerned.

In order to address this problem, what you basically have to do is to split the flash multiple slots and update a slot other than the current one.

The ESP8266 CPU can memory map only 1Mb of flash. This 1Mb window can be moved in 1Mb increments and thus we can execute the firmware linked at a given absolute address if each firmware slot is spaced at 1Mb increments. That's why we limit the size of the FW+FS bundle to 1Mb.

You can try to tweak that hardcoded 0x100000 is FS_ADDR so that a larger filesystem can be positioned correctly. Note that OTA won't work in that case.

alashkin, 2016-08-10:
@mmikulicic 

Also I'm not sure MMAP will work

alashkin, 2016-08-10:
what _will_ work is a usage of 2 FSs
1 - ususal, system
2 - mounted with SPIFFS_mount - any size, any place

wallaroo, 2016-08-10:
so... if i understood following the solution #2 of @alashkin in my case (4M flash size) should be possible to layout flash with 2 slots of 1M for FW and a slot of 2M for the second FS right?
in this way ota should still work?

mmikulicic, 2016-08-10:
yeah, Alex is correct, mmapping wouldn't work outside the current 1Mb window.

You can use a second FS image and put it where you want.

If you add it in the manifest.json file, MFT flash tool will do the right thing.

Note that currently our f* function shims (fopen, fread etc) assumes the first filesystem. With some amount of tweaking you can make them point to your second filesystem. It really depends what you want to do.

wallaroo, 2016-08-10:
cool! i'll try to do something of reusable... so maybe a pull request could arrive ;-)

wallaroo, 2016-08-11:
hi! i was preparing the pull request for the FS_EXT but after a git pull i can't build... i get this error:
```
root@aa3df25dacf7:~/mongoose# make -C fw/platforms/esp8266/
make: Entering directory '/root/mongoose/fw/platforms/esp8266'
Unable to find image 'docker.cesanta.com:5000/esp8266-build-oss:2.0.0-r1' locally
docker: Error response from daemon: Get https://docker.cesanta.com:5000/v1/_ping: dial tcp 104.155.75.84:5000: getsockopt: connection refused.
See 'docker run --help'.
Makefile:24: recipe for target 'all' failed
make: *** [all] Error 125
make: Leaving directory '/root/mongoose/fw/platforms/esp8266'
root@aa3df25dacf7:~/mongoose#
```

some suggestion?

mmikulicic, 2016-08-12:
@wallaroo, do you have any firewall that blocking access to port 5000 ?

FWIW, we actually serve the images also on port 443, but didn't yet change all the references in the repository (turns out there are some yaks to shave in our continuous integration infrastructure).

So technically you can pull `docker.cesanta.com/esp8266-build-oss:2.0.0-r1` and use `docker tag` to locally rename it as `docker.cesanta.com:5000/esp8266-build-oss:2.0.0-r1`.

cpq, 2016-12-27:
Closing this.

closed 2016-12-27 by cpq