cesanta/mongoose-os #153: PWM values for controlling servo

carterw, 2016-09-04:
I'm trying to control a servo with the PWM function described in the API, and it is not working properly for any of the period/duty values I've tried.

https://www.servocity.com/how-does-a-servo-work

" The servo expects to see a pulse every 20 ms. The length of the pulse will determine how far the motor turns. For example, a 1.5 ms pulse will make the motor turn to the 90 degree position (neutral position)."

I'm assuming the 'period' value in the API is the frequency?  What pairs of values will result in pulse widths as follows;

Position "0" (1.5ms pulse) is middle, "90" (~2ms pulse) is all the way to the right, "-90" (~1ms pulse) is all the way to the left.

carterw, 2016-09-06:
Looking in the source code I am seeing the file esp_pwm.c, is this what generates the pulse train? I don't have easy access to an oscilloscope, did anyone ever verify the PWM behavior of that code?

carterw, 2016-09-08:
No response on this? Correct PWM behavior is essential for many applications.

carterw, 2016-09-10:
It turns out to just be a matter of arithmetic. :)

This yields a 1ms pulse train with a frequency of 20ms;
PWM.set(14, 20000, 1000)

This yields a 2ms pulse train;
PWM.set(14, 20000, 2000)

It turns out that I can operate small servo motors (TowerPro SG92R) with these pulses at 3.3V (both power and signal), which is very darn handy.

cpq, 2017-02-09:
hi @carterw ,
FYI we have done with the JS engine refactoring.
Are you still using mOS ?

carterw, 2017-02-09:
Thanks for the info. When you dropped your previous JavaScript API I began working with Micropython, and have had fair success with it. Are there docs that describe your new implementation? If so please post a link and I will have a look.

cpq, 2017-02-09:
intro: https://mongoose-os.com/blog/mjs-a-new-approach-to-embedded-scripting/

To try:
```
mos flash --firmware https://mongoose-os.com/downloads/mos-esp8266.zip
mos get init.js
mos put init.js
mos console
```

rohithreddy, 2017-02-11:
hi @carterw have you got the SG90 working with mongoose . if so need you help regarding it

cpq, 2017-02-13:
Hi @rohithreddy !
PWM is only available on esp8266 currently. If you want to use it, do
1. checkout mongoose-os repo
2. cd to fw/examples/mjs_base
3. add line "mgos_pwm_set" to the list of exported symbols
4. type "mos build --arch esp8266"
5. type "mos flash"
6. Now you can use PWM from JS: `let PWM = { set: ffi('int mgos_pwm_set(int, int, int)') }; `

rohithreddy, 2017-02-14:
Thanks @cpq , will let you know how it works out.

cpq, 2017-02-17:
ok, closing this.

closed 2017-02-17 by cpq

ticklemynausea, 2017-04-20:
Hi @cpq 

Could you please clarify step 3, "add line "mgos_pwm_set" to the list of exported symbols" ?

Thank you.

tsofist, 2017-05-20:
I have error:
``MJS error: bad ffi signature: "int mgos_pwm_set(int, int, int)": dlsym('mgos_pwm_set') failed``

Can anyone suggest what I'm doing wrong?
I'm not strong in C =(

cpq, 2017-05-23:
please follow https://forum.mongoose-os.com/discussion/758/pwm-how-to-with-esp8266-need-some-tips-please#latest