cesanta/mongoose-os #154: recent flash broken?

carterw, 2016-09-16:
I cloned and built 8266 FW today, flashed it to a device and I am not getting a prompt to the MFT console. After reset this is printed and then no response;

Mongoose IoT Firmware 20160916-221531/master@3d03aac6
esp_sj_init          SDK 2.0.0(656edbf), RAM: 48704 total, 44004 free
esp_print_reset_info Reset cause: 6 (sys reset)
fs_mount             Mounting FS: 131072 @ 0xdb000
sj_sys_config_init   MAC: 5ECF7F88B628
clubby_channel_ws_out_connect 0x3fff17e4 Connecting to api.mongoose-iot.com, SSL? 0clubby_channel_uart  0x3fff1f7c UART0
"""

carterw, 2016-09-16:
Due to a commit yesterday;
Turn off JS by default; enable Clubby on UART0 

Wow, where are you guys going with this?

cpq, 2016-09-17:
Couple of things is going to happen.

1. We're sunsetting JS. Rationale: porting to boards that are too small for JS
2. We're getting rid of cloud IDE and instead providing a cmd line tool for building/flashing FW. Rationale: people mostly don't use online IDEs, they use their own tools/environment they . A command line utility integrates smoothly in any existing process (editor, IDE, version control, etc), unlike an online IDE.
3. We're closing FW sources, and opening them only for paid users.

carterw, 2016-09-17:
Very sorry to hear this. Being able to code in JavaScript is the one thing that has made your platform interesting to me. And even then, I have been forced to incorporate most of my code into the FW due to the unfortunate need to reserve about half of available memory for JS strings. I'll stick with the FW I am currently able to build and begin investigating other platforms.

Eliminating JS and closing the FW sources is an abandonment of independent developers like myself (assuming there actually are any others). I realize this is a most likely a business decision, you get no revenue from people like me, but it is a little startling to find out about it so abruptly.

realzeus, 2016-09-17:
Your JS implementation was just perfect. It was possible to develop with speed of lightning.
I'm really sorry to hear you are sunsetting it.
For now I will continue to work with the clone of your repository.

cpq, 2016-09-18:
@realzeus could you elaborate on that please?
How your development changed with MIOT - how it was before and how it is now?
What sort of projects/functionality you're working on?

That can help just justify not closing the source and not deleting JS.
Right now, we're under impression that JS features and open sources just do not worth our efforts.

Thanks!

wallaroo, 2016-09-19:
hi guys, i think js is what differentiates this platform from the others. however, I am one of those who doesn't use web ide... i think CLI is a great idea and maybe even a plugin for the most popular ide (Eclipse, Jetbrains WS/IJ). The ability to run JS on IoT devices enables a great community of developers and amateurs to create applications...
like explained in https://github.com/cesanta/mongoose-iot/pull/152#issuecomment-240028802 i think there is a problem with FS space to proficiently use js... but i guess it is solvable with some tuning with the using of flash during OTA... (atleast for ESP)
...so if JS will be dropped out, i think i'll move to other platforms... (i know, i'm a drop in the ocean).

Michele

realzeus, 2016-09-19:
I found your platform perfect because it's not monolithic, like Arduino and many others. On that platforms you have to compile everything first.
Contrary, your platform looks like SOA, where C code provides stable micro-services and JS code works like orchestrator.
With minimal development I was able to send JS code over WLAN and execute it even without saving to file system. This approach has drastically reduced the time of development cycle.

Arduino change cycle: (compile + upload + restart) is approximately 20 seconds.
ESP with MIOT: (upload, compilation and execution is automatic) takes less than 1 second.

I use ESP with MIOT in 4 projects:
- internet connected temperature sensors for energy consumption optimization (DHT22)
- presentation stand controlled over internet: http://www.iot40systems.com/en/caberra.html#iotlive
- bike tracking (ESP is used as controller only, WLAN of ESP is not used, connection over GPRS)
- remote control for toy vehicle

Thank you very much for your work!

cpq, 2016-09-21:
Gents, thanks for your feedback. It is very much appreciated.

@realzeus , your reasoning is almost word-to-word our reasoning a year ago, when we started working on MIOT.

A year has passed and our vision did not change. We believe that the future is in embedded scripting , and JS has been chosen as a language which is not perfect for embedded, but so widely used that it is justified.

Said that, we believe there is still some time needed to accept that widely.

After the extensive discussion, our strategy is to keep the JS support, but de-prioritize it, and turn it off by default. We are full speed on making our CLI tool work nicely with the FW, and we're quite close with making it work with our public managed service.

Bottom line: JS is off by default, FW sources stay open, `miot` CLI tool is on the way.

My apologies for having things broken sometimes: the platform is a huge thing for a small team like ours (the JS engine itself worth a big project).

Would you guys be willing to pay a small subscription fee for using MIOT and keep us working on it - developing device provisioning, mobile SDKs, access management, better JS support, dashboard builders, etc etc ?

realzeus, 2016-09-21:
Thank you for these decisions!
Most important is what it stays open.
And I really hope to contribute sooner or later.

I think, it would make sense to enable JS on Platforms, which have enough resources, and disable it on more restricted platforms.

I'm personally not sure about subscription - I don't like constant obligations.
I will be happy to make a 1-time payment on every successful project.

carterw, 2016-09-21:
The JavaScript feature is the only thing I care much about so I can't justify paying a subscription, sorry. If necessary I can move to Lua, Espruino, or MicroPython which seem to be at a similar stage of development. 

It would be a shame, because I have stayed with you for many months as bugs have slowly been fixed. Only recently am I able to make multiple devices perform commonplace IoT functions in response to server commands with fair reliability.

carterw, 2016-11-09:
I tried cloning and building fresh firmware today. Previously I could enable JavaScript by editing Makefile.build and setting;
MG_ENABLE_JS ?= 1

Building and flashing with this flag now does not produce a prompt? But I do see;
"Below you should see a JS prompt, feel free to play with it."

Is there an additional step?

cpq, 2016-11-09:
Hi Bill, we're making a change to the build system, by the end of today we expect this to work.

carterw, 2016-11-09:
There are several /fs directories, which one? And will files placed there be included and precompiled in flash memory? My small program is too large to load into runtime RAM.

Also, what do I need to do in order to get a prompt on the MFT console?

cpq, 2016-11-10:
@carterw , please don't use MFT.
Follow https://mongoose-iot.com/docs/#/quickstart/overview.md/
At this particular moment, `miot build` does not build JS-enabled firmwares, but that is going to change, hopefully, today, because we're working on it at this precise moment.

carterw, 2016-11-10:
So all custom source code intended to be resident on the device has to be passed along to your server in order to be included in a firmware build? I hope I am misunderstanding because that seems invasive to me, why should I have to give you my source code? Also I am not seeing any mention of a console or any debug mechanism.

cpq, 2016-11-10:
1. `miot console` does the job.
2. `miot build --local` does the local build, needs cesanta/mongoose-iot repo checked out.

With (2), the source code is not sent to us, but one needs to clone the repo.
Please update your `miot` binary if it does not understand `--local` yet. Note that this functionality is quite fresh, so you might experience some hiccups.

carterw, 2016-11-11:
Glad to hear there is a local build option, thanks. It sounds like I should wait until I hear from you that the JavaScript build option is fully in place.

cpq, 2016-11-15:
JS support is in the build infrastructure - https://github.com/cesanta/mongoose-iot/tree/master/fw/examples/js_hello

Note that we are working on the build system and JS support itself very actively, so changes are expected. I am closing this issue.

closed 2016-11-15 by cpq