cesanta/mongoose-os #157: build failing

wallaroo, 2016-10-12:
hi! i'm trying to build the master with following configuration in fw/platform/esp8266/Makefile.build :
```
[...]
MG_ENABLE_JS ?= 1
MG_ENABLE_CLUBBY ?= 0
MG_ENABLE_CLUBBY_UART ?= 0
MG_ENABLE_CONFIG_SERVICE ?= 0
MG_ENABLE_FILESYSTEM_SERVICE ?= 1
MG_ENABLE_UPDATER_POST ?= 1
MG_ENABLE_UPDATER_CLUBBY ?= 0
MG_ENABLE_DNS_SD ?= 1
[...]
V7_FREEZE ?= 1
[...]
```
and i'm getting this error:
```
make: Entering directory '/root/cesanta/mongoose-iot/fw/platforms/esp8266'
make: Entering directory '/src/fw/platforms/esp8266'
make -j4 -C /src//fw/platforms/posix \
  REPO_PATH=/src/ \
  V7_FEATURES='-DV7_ENABLE__File__list=1 -DV7_ENABLE__File__require=1 -DV7_ENABLE__Function__bind=1 -DV7_ENABLE__Function__call=1 -DV7_ENABLE__Math=1 -DV7_ENABLE__Math__ceil=1 -DV7_ENABLE__Math__floor=1 -DV7_ENABLE__Math__max=1 -DV7_ENABLE__Math__min=1 -DV7_ENABLE__Math__random=1 -DV7_ENABLE__Math__round=1 -DV7_ENABLE__Memory__stats=1 -DV7_ENABLE__Proxy=1 -DV7_BUILD_PROFILE=1 -DV7_ENABLE_FILE -DV7_DISABLE_AST_TAG_NAMES -DV7_ENABLE__RegExp=1 -DV7_ENABLE__StackTrace=1 -DV7_ENABLE__Object__getPrototypeOf -DV7_ENABLE__Object__isPrototypeOf -DV7_ENABLE__Object__hasOwnProperty -DV7_ENABLE__Object__keys -DV7_ENABLE__Object__isExtensible -DV7_EXPOSE_PRIVATE -DV7_ENABLE_ENTITY_IDS -DV7_MAIN -DV7_FREEZE' \
  BIN_DIR=/src/fw/platforms/esp8266/.build/posix_sjs_bin \
  BUILD_DIR=/src/fw/platforms/esp8266/.build/posix_sjs_build
make[1]: Entering directory '/src/fw/platforms/esp8266'
make[1]: warning: -jN forced in submake: disabling jobserver mode.
Makefile:56: warning: undefined variable 'MG_ENABLE_JS'
make[1]: Circular /src/fw/platforms/esp8266/.build/posix_sjs_build/build_info.c <- /src/fw/platforms/esp8266/.build/posix_sjs_build/build_info.o dependency dropped.
LD /src/fw/platforms/esp8266/.build/posix_sjs_bin/mongoose-iot
MKFS fs
/src/fw/platforms/esp8266/.build/posix_sjs_build/mg_mdns.o: In function `on_wifi_change':
/src/fw/platforms/posix/../../src/mg_mdns.c:60: undefined reference to `mg_mdns_hal_join_group'
collect2: error: ld returned 1 exit status
Makefile:237: recipe for target '/src/fw/platforms/esp8266/.build/posix_sjs_bin/mongoose-iot' failed
make[1]: *** [/src/fw/platforms/esp8266/.build/posix_sjs_bin/mongoose-iot] Error 1
make[1]: Leaving directory '/src/fw/platforms/posix'
Makefile.build:476: recipe for target '/src/fw/platforms/esp8266/.build/posix_sjs_bin/mongoose-iot' failed
make: *** [/src/fw/platforms/esp8266/.build/posix_sjs_bin/mongoose-iot] Error 2
make: Leaving directory '/src/fw/platforms/esp8266'
../../../fw/docker_common.mk:59: recipe for target 'all' failed
make: *** [all] Error 2
make: Leaving directory '/root/cesanta/mongoose-iot/fw/platforms/esp8266'
```

there is some dependency through MG modules?
many thanks!

wallaroo, 2016-10-12:
i added the MG_ENABLE_JS parameter copy that seems to be missing

```
$(MIOT_POSIX_BIN):
	$(MAKE) -j4 -C $(MIOT_POSIX_PATH) \
	  REPO_PATH=$(REPO_PATH) \
	  MG_ENABLE_JS=$(MG_ENABLE_JS) \
	  V7_FEATURES='$(V7_FEATURES) -DV7_MAIN -DV7_FREEZE' \
	  BIN_DIR=$(MIOT_POSIX_BIN_DIR) \
	  BUILD_DIR=$(MIOT_POSIX_BUILD_DIR)
endif
```
but i've just changed the error:
```
make: Entering directory '/root/mongoose-iot/fw/platforms/esp8266'
make: Entering directory '/src/fw/platforms/esp8266'
make -j4 -C /src//fw/platforms/posix \
  REPO_PATH=/src/ \
  MG_ENABLE_JS=1 \
  V7_FEATURES='-DV7_ENABLE__File__list=1 -DV7_ENABLE__File__require=1 -DV7_ENABLE__Function__bind=1 -DV7_ENABLE__Function__call=1 -DV7_ENABLE__Math=1 -DV7_ENABLE__Math__ceil=1 -DV7_ENABLE__Math__floor=1 -DV7_ENABLE__Math__max=1 -DV7_ENABLE__Math__min=1 -DV7_ENABLE__Math__random=1 -DV7_ENABLE__Math__round=1 -DV7_ENABLE__Memory__stats=1 -DV7_ENABLE__Proxy=1 -DV7_BUILD_PROFILE=1 -DV7_ENABLE_FILE -DV7_DISABLE_AST_TAG_NAMES -DV7_ENABLE__RegExp=1 -DV7_ENABLE__StackTrace=1 -DV7_ENABLE__Object__getPrototypeOf -DV7_ENABLE__Object__isPrototypeOf -DV7_ENABLE__Object__hasOwnProperty -DV7_ENABLE__Object__keys -DV7_ENABLE__Object__isExtensible -DV7_EXPOSE_PRIVATE -DV7_ENABLE_ENTITY_IDS -DV7_MAIN -DV7_FREEZE' \
  BIN_DIR=/src/fw/platforms/esp8266/.build/posix_sjs_bin \
  BUILD_DIR=/src/fw/platforms/esp8266/.build/posix_sjs_build
make[1]: Entering directory '/src/fw/platforms/esp8266'
make[1]: warning: -jN forced in submake: disabling jobserver mode.
make[1]: Circular /src/fw/platforms/esp8266/.build/posix_sjs_build/build_info.c <- /src/fw/platforms/esp8266/.build/posix_sjs_build/build_info.o dependency dropped.
LD /src/fw/platforms/esp8266/.build/posix_sjs_bin/mongoose-iot
MKFS fs
/src/fw/platforms/esp8266/.build/posix_sjs_build/v7.o: In function `mbuf_resize':
/src/fw/platforms/posix/../../../v7/v7.c:7332: multiple definition of `mbuf_resize'
/src/fw/platforms/esp8266/.build/posix_sjs_build/mongoose.o:/src/fw/platforms/posix/../../../mongoose/mongoose.c:999: first defined here
/src/fw/platforms/esp8266/.build/posix_sjs_build/v7.o: In function `mbuf_init':
/src/fw/platforms/posix/../../../v7/v7.c:7320: multiple definition of `mbuf_init'
/src/fw/platforms/esp8266/.build/posix_sjs_build/mongoose.o:/src/fw/platforms/posix/../../../mongoose/mongoose.c:987: first defined here
/src/fw/platforms/esp8266/.build/posix_sjs_build/v7.o: In function `mbuf_free':
/src/fw/platforms/posix/../../../v7/v7.c:7326: multiple definition of `mbuf_free'
/src/fw/platforms/esp8266/.build/posix_sjs_build/mongoose.o:/src/fw/platforms/posix/../../../mongoose/mongoose.c:993: first defined here
/src/fw/platforms/esp8266/.build/posix_sjs_build/v7.o: In function `mbuf_trim':
/src/fw/platforms/posix/../../../v7/v7.c:7347: multiple definition of `mbuf_trim'
/src/fw/platforms/esp8266/.build/posix_sjs_build/mongoose.o:/src/fw/platforms/posix/../../../mongoose/mongoose.c:1014: first defined here
/src/fw/platforms/esp8266/.build/posix_sjs_build/v7.o: In function `mbuf_insert':
/src/fw/platforms/posix/../../../v7/v7.c:7353: multiple definition of `mbuf_insert'
/src/fw/platforms/esp8266/.build/posix_sjs_build/mongoose.o:/src/fw/platforms/posix/../../../mongoose/mongoose.c:1020: first defined here
/src/fw/platforms/esp8266/.build/posix_sjs_build/v7.o: In function `mbuf_append':
/src/fw/platforms/posix/../../../v7/v7.c:7383: multiple definition of `mbuf_append'
/src/fw/platforms/esp8266/.build/posix_sjs_build/mongoose.o:/src/fw/platforms/posix/../../../mongoose/mongoose.c:1050: first defined here
/src/fw/platforms/esp8266/.build/posix_sjs_build/v7.o: In function `mbuf_remove':
/src/fw/platforms/posix/../../../v7/v7.c:7387: multiple definition of `mbuf_remove'
/src/fw/platforms/esp8266/.build/posix_sjs_build/mongoose.o:/src/fw/platforms/posix/../../../mongoose/mongoose.c:1054: first defined here
/src/fw/platforms/esp8266/.build/posix_sjs_build/v7.o: In function `c_strnlen':
/src/fw/platforms/posix/../../../v7/v7.c:7409: multiple definition of `c_strnlen'
/src/fw/platforms/esp8266/.build/posix_sjs_build/mongoose.o:/src/fw/platforms/posix/../../../mongoose/mongoose.c:1397: first defined here
/src/fw/platforms/esp8266/.build/posix_sjs_build/v7.o: In function `c_vsnprintf':
/src/fw/platforms/posix/../../../v7/v7.c:7468: multiple definition of `c_vsnprintf'
/src/fw/platforms/esp8266/.build/posix_sjs_build/mongoose.o:/src/fw/platforms/posix/../../../mongoose/mongoose.c:1456: first defined here
/src/fw/platforms/esp8266/.build/posix_sjs_build/v7.o: In function `c_snprintf':
/src/fw/platforms/posix/../../../v7/v7.c:7605: multiple definition of `c_snprintf'
/src/fw/platforms/esp8266/.build/posix_sjs_build/mongoose.o:/src/fw/platforms/posix/../../../mongoose/mongoose.c:1593: first defined here
/src/fw/platforms/esp8266/.build/posix_sjs_build/v7.o: In function `c_strnstr':
/src/fw/platforms/posix/../../../v7/v7.c:7645: multiple definition of `c_strnstr'
/src/fw/platforms/esp8266/.build/posix_sjs_build/mongoose.o:/src/fw/platforms/posix/../../../mongoose/mongoose.c:1633: first defined here
/src/fw/platforms/esp8266/.build/posix_sjs_build/v7.o: In function `cs_base64_init':
/src/fw/platforms/posix/../../../v7/v7.c:9084: multiple definition of `cs_base64_init'
/src/fw/platforms/esp8266/.build/posix_sjs_build/mongoose.o:/src/fw/platforms/posix/../../../mongoose/mongoose.c:319: first defined here
/src/fw/platforms/esp8266/.build/posix_sjs_build/v7.o: In function `cs_base64_update':
[... cut ...]
make[1]: *** [/src/fw/platforms/esp8266/.build/posix_sjs_bin/mongoose-iot] Error 1
make[1]: Leaving directory '/src/fw/platforms/posix'
Makefile.build:498: recipe for target '/src/fw/platforms/esp8266/.build/posix_sjs_bin/mongoose-iot' failed
make: *** [/src/fw/platforms/esp8266/.build/posix_sjs_bin/mongoose-iot] Error 2
make: Leaving directory '/src/fw/platforms/esp8266'
../../../fw/docker_common.mk:59: recipe for target 'all' failed
make: *** [all] Error 2
make: Leaving directory '/root/mongoose-iot/fw/platforms/esp8266'
```

dimonomid, 2016-10-12:
Thanks for reporting that.

Yes I checked and indeed freezing doesn't work properly at the moment; we should either add a test for it or drop it.

I was able to build the firmware with `V7_FREEZE` set to 0 and `MG_ENABLE_UPDATER_CLUBBY` set to 1. We'll write here once we resolve it.

closed 2016-10-12 by cesantabot