cesanta/mongoose-os #160: CoAP memory leak

iliapenev, 2016-11-03:
There is memory leak, when sending CoAP messages.
Demo code could be found here:
https://gist.github.com/anonymous/61d7360c0a27eb2c4922b50ffb2be1ca
It's based on c_hello project.

```
coap_send_message      50129 Sending 1257
umm_oom_cb             50100 Failed to allocate 34
coap_send_message       4440 Sending 1258
umm_oom_cb 46619 Failed to allocate 16
```

When "Failed to allocate" appears, messages are recieved without CoAP options.
Possibly problem is in handling CoAP options.
Wireshark dump: https://drive.google.com/file/d/0B4Kd4pQQfRXeYTNDbkYyMDIwT3M/view?usp=sharing

alashkin, 2016-11-03:
@iliapenev 
As far as I can see `mg_coap_free_options` is missed after `	mg_coap_send_message(coap_connection, &msg);`

`mg_coap_add_option` allocates options by using `alloc()`; Mongoose doesn't automatically free options  to allow resend the same message if timeout expired (UDP :( )

iliapenev, 2016-11-03:
Sorry, I missed that. 
Resend should be handled by user, as I understand?

alashkin, 2016-11-03:
>Resend should be handled by user, as I understand?

Yes :(

alashkin, 2016-11-03:
Anyway, I'll update doc

iliapenev, 2016-11-03:
Sorry for inconvenience. 
Closing issue.

closed 2016-11-03 by iliapenev

alashkin, 2016-11-03:
No worries! Thank you for feedback, I've updated doc