cesanta/mongoose-os #161: How I Build and Suggestions

ser007, 2016-11-08:
Great Design! Thank you for sharing the experience and points of view.

I want to learn more about the codes.

What operating system, IDE, and compiler do I use to develop and build?

Any suggestions for wanting to learn more about the codes?

Thanks. Andréa

cpq, 2016-11-08:
Thanks Andréa!
Could you be more specific please?

ser007, 2016-11-08:
Hi @cpq,

I watched the video https://www.youtube.com/watch?v=kThcRN8VF0Q and I was interested in the capabilities you implemented in a esp8266.

So I'm thinking that if I start studying I get to understand how the firmware was developed.

I have a knowledge in javascript and I use visual studio code on windows in my work. I have never developed in C / C ++, I have no knowledge of the language, nor the compiler, the best operating system to compile and not the best.

So my question. So what advice can you give me to start this way?

Thanks. Andréa

cpq, 2016-11-08:
Thank you.
So, do you want to develop device logic in JavaScript, or you'd like to learn C/C++ for that?

ser007, 2016-11-08:
I want to learn C / C ++, books you recommend, documentation, and how mongoose-iot works, which operating system to mount on my virtual machine, which compiler and IDE.

cpq, 2016-11-08:
Sure.
Please start with this:
https://mongoose-iot.com/docs/#/quickstart/

As for the book on C, I'd recommend a classic - Kernighan & Ritchie, "The C Programming Language".

Thank you!

closed 2016-11-08 by cpq

ser007, 2016-11-19:
Hi @cpq!

Excellent work from you!

Great documentation, I started to read.

Thanks for the book, I downloaded it.

Thank you!