cesanta/mongoose-os #162: ESP8266 save_cfg

iliapenev, 2016-11-10:
I'm trying to save to config on ESP8266.
Result is : 
Trap 29: pc=0x40211438 va=0x0
Dumping core
--- BEGIN CORE DUMP ---
{"arch": "ESP8266", "cause": 29, "REGS": {"addr": 1073675620, "data": "NhQhQED+/z9A/v8/AAAAANQAAAAU//8/EP//Pw0AAABUhP4/AAEAAAAAAAAAgP//AAAAAAAAAADw/f4/AAYAYDgUIUAfAAAAAAAAAAAAAAAAAAAAMAAAAA=="}
, "DRAM": {"addr": 1073643520, "data": "jJT+

Sample code here:
https://gist.github.com/anonymous/cfc9f43aef644707d588a48183a07245

Tested with mongoose-iot c7defce77963c41e128195c6a28423cedb476ec1

iliapenev, 2016-11-11:
Here you can find log from console. 
Could you please help @alashkin @rojer @cpq 
https://gist.github.com/anonymous/7229317446bc1dd6fdce03561c76f3ad

rojer, 2016-11-11:
msg, the second parameter, can't be NULL.
char *error_message;
save_cfg(cfg, &error_message);

iliapenev, 2016-11-11:
save_cfg works with NULL for error_message on CC3200. 

With parameter now is fine. 
But there is   
set_errno               7919 spiffs error: -10002

`mg_conf_parse_cb        5936 Set [sys.wdt_timeout] = [15]
mg_conf_parse_cb        4927 Set [conf_acl] = [*]
mg_conf_parse_cb        4417 Set [hello.who] = [world]
mg_conf_parse_cb        4626 Set [hello.val] = [0]
load_config_file        4412 === Loading conf_vendor.json
set_errno               7919 spiffs error: -10002
save_cfg               22489 Saved to conf.json
----------
 Counter value= 10!
----------
load_config_file      984945 === Loading conf_defaults.json
mg_conf_parse_cb        5117 Set [wifi.sta.enable] = [false]
mg_conf_parse_cb         467 Set [wifi.sta.ssid] = []
`

Could you please check other issues. 
We are trying to migrate CC3200 project to ESP8266

rojer, 2016-11-11:
the first thing save_cfg does is set *err_msg to NULL.

`save_cfg works with NULL for error_message on CC3200.`
that's because CC3200 is magic - NULL dereferences do not trigger exceptions. internal ROM is mapped at address 0, so reading it is obviously fine, but so is writing, which is unexpected.

wrt log you posted - spiffs errors are benign, conf_vendor.json does not exist and it's ok.
it says it saved the config, so it's fine.

rojer, 2016-11-14:
i'm going to close this - looks like a NULL dereference in your caught that was not caught on CC3200 because it's weird.

closed 2016-11-14 by rojer

iliapenev, 2016-11-14:
Thanks!
I've missed double ** in save_cfg() params.