cesanta/mongoose-os #164: ESP8266 MD5

iliapenev, 2016-11-10:
In my project I'm using cs_md5() function.
Compile gives error:
```
/app/.build/mongoose-iot.a(mongoose.o):(.text.cs_md5+0x0): undefined reference to `MD5_Init'
/app/.build/mongoose-iot.a(mongoose.o):(.text.cs_md5+0x4): undefined reference to `MD5_Update'
/app/.build/mongoose-iot.a(mongoose.o):(.text.cs_md5+0x8): undefined reference to `MD5_Final'
/app/.build/mongoose-iot.a(mongoose.o): In function `cs_md5':
/mongoose-iot/mongoose/mongoose.c:12187: undefined reference to `MD5_Init'
/mongoose-iot/mongoose/mongoose.c:12187: undefined reference to `MD5_Update'
/mongoose-iot/mongoose/mongoose.c:12187: undefined reference to `MD5_Final'
```
Removing -DDISABLE_MD5 fixs problem.
https://github.com/cesanta/mongoose-iot/blob/master/fw/platforms/esp8266/Makefile.build#L207

rojer, 2016-11-14:
there is an MD5 implementation in ROM, we use it for SSL.
https://github.com/cesanta/mongoose-iot/blob/master/common/platforms/esp8266/esp_crypto.c

closed 2016-11-14 by rojer

iliapenev, 2016-11-14:
@rojer 
problem still stays. 
See example. Directly cs_md5 is not used. 
https://gist.github.com/anonymous/a605d995ab9feee80a2142019a42addf