cesanta/mongoose-os #175: mongoose iot cloud

mark-hahn, 2016-12-26:
I know this question is not a technical one for this repo, but I couldn't find a forum.  The mongoose IOT cloud looks very interesting.  It is not described in detail anywhere I can find.  Is it free?  Is the cloud-building feature a separate service than the IOT data collection and control?

cpq, 2016-12-27:
@mark-hahn where did you look for the information?
I'd like to stick links to the forum (forum.cesanta.com) to the appropriate place.
Mongoose is free, open source, under the dual license (GPLv2 and commercial).

closed 2016-12-27 by cpq

mark-hahn, 2016-12-27:
I looked on every page I could find in the website.

 > Mongoose is free, open source,

I know.  I was looking for cloud information.  To repeat:

The mongoose IOT cloud  ...  Is it free? Is the cloud-building feature a separate service than the IOT data collection and control?