cesanta/mongoose-os #176: esp8266: crash during ssl handshake

kzyapkov, 2016-12-30:
Hi all,

We've observed a number of instances of this, here is one:

```
(gdb) bt full
#0  ssl_socket_recv (ctx=0x3fff7cfc, buf=0x3fff7e4c "", len=5)
    at /mongoose-iot/mongoose/mongoose.c:14628
        nc = 0x3fff7cfc
        cs = 0x3fff5e64
        seg = 0x800
        __func__ = "ssl_socket_recv"
        seg_len = <optimized out>
#1  0x402449c3 in mbedtls_ssl_fetch_input (ssl=ssl@entry=0x3fff7c04, 
    nb_want=nb_want@entry=5) at ssl_tls.c:2513
        ret = <optimized out>
        len = 5
        want_buf_size = 13
#2  0x40245e70 in mbedtls_ssl_read_record_layer (ssl=ssl@entry=0x3fff7c04)
    at ssl_tls.c:3920
        ret = <optimized out>
#3  0x402466e6 in mbedtls_ssl_read_record (ssl=ssl@entry=0x3fff7c04)
    at ssl_tls.c:3865
        ret = <optimized out>
#4  0x402484d5 in ssl_parse_server_hello (ssl=0x3fff7c04) at ssl_cli.c:1456
        buf = <optimized out>
        t = <optimized out>
        ret = <optimized out>
        n = <optimized out>
        ext_len = <optimized out>
        comp = <optimized out>
        i = <optimized out>
        ext = <optimized out>
        handshake_failure = 0
        suite_info = <optimized out>
#5  mbedtls_ssl_handshake_client_step (ssl=ssl@entry=0x3fff7c04)
    at ssl_cli.c:3416
        ret = <optimized out>
#6  0x40244d5c in mbedtls_ssl_handshake_step (ssl=ssl@entry=0x3fff7c04)
    at ssl_tls.c:6470
        ret = -28800
#7  0x40244dc4 in mbedtls_ssl_handshake (ssl=0x3fff7c04) at ssl_tls.c:6494
        ret = 0
#8  0x4021e2be in mg_ssl_if_handshake (nc=nc@entry=0x3fff7cfc)
    at /mongoose-iot/mongoose/mongoose.c:4490
        ctx = 0x3fff5e84
---Type <return> to continue, or q <return> to quit---
        err = <optimized out>
#9  0x40225014 in mg_lwip_ssl_do_hs (nc=nc@entry=0x3fff7cfc)
    at /mongoose-iot/mongoose/mongoose.c:14478
        cs = 0x3fff5e64
        server_side = <optimized out>
        res = <optimized out>
        __func__ = "mg_lwip_ssl_do_hs"
#10 0x402252c8 in mg_ev_mgr_lwip_process_signals (
    mgr=mgr@entry=0x3fff4550 <s_mgr>) at /mongoose-iot/mongoose/mongoose.c:14287
        nc = 0x3fff7cfc
        cs = 0x3fff5e64
        md = 0x3fff50dc
#11 0x402254f5 in mg_lwip_if_poll (iface=<optimized out>, 
    timeout_ms=<optimized out>) at /mongoose-iot/mongoose/mongoose.c:14353
        mgr = 0x3fff4550 <s_mgr>
        n = 0
        nc = <optimized out>
        tmp = <optimized out>
        min_timer = 0
        num_timers = 0
#12 0x4021dca2 in mg_mgr_poll (m=m@entry=0x3fff4550 <s_mgr>, 
    timeout_ms=timeout_ms@entry=0) at /mongoose-iot/mongoose/mongoose.c:2279
        i = 0
        now = <optimized out>
        __func__ = "mg_mgr_poll"
#13 0x40219d8a in mongoose_poll (ms=ms@entry=0)
    at /mongoose-iot/fw/src/miot_mongoose.c:52
        ret = <optimized out>
        __func__ = "mongoose_poll"
#14 0x40226c8b in miot_lwip_task (e=<optimized out>)
    at /mongoose-iot/fw/platforms/esp8266/user/esp_task.c:82
        timeout_ms = <optimized out>
        mgr = 0x3fff4550 <s_mgr>
#15 0x40000f49 in ?? ()
No symbol table info available.
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) print ((struct mg_connection *) ctx)->recv_mbuf
$33 = {buf = 0x0, len = 0, size = 0}
(gdb) 
```

Not sure if the crash happens before the recv() invocation or after. These usually happen immediately after re-associating to an AP while in STA mode. Here is the device log immediately preceding the crash:

```
cloud_write_handler  Bytes written: 534
coiot_switch_status  one is on and true
logmem               heap: 18460 / 30248; 190.87 ticks/s
logmem               heap: 18460 / 30248; 191.97 ticks/s
coiot_switch_status  one is on and true
state: 5 -> 2 (2c0)
rm 0
pm close 7
miot_wifi_on_change_cb Wifi: disconnected
_wifi_change_cb      WiFi status=2; mode=1, STA=1 AP=0
cloud_close_handler  Cloud connection closed
reconnect
state: 2 -> 0 (0)
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with oh'shelly, channel 1
dhcp client start...
miot_wifi_on_change_cb Wifi: connected
_wifi_change_cb      WiFi status=0; mode=1, STA=1 AP=0
ip:192.168.42.140,mask:255.255.255.0,gw:192.168.42.1
miot_wifi_on_change_cb WiFi: ready, IP 192.168.42.140
_wifi_change_cb      WiFi status=2; mode=1, STA=1 AP=0
cloud_connect        Connect to dispatcher tcp://our.iot.backend:6010
miot_mdns_hal_join_group Joining multicast group 224.0.0.251

Trap 28: pc=0x4021be15 va=0x80a
Dumping core
```

rojer, 2016-12-30:
thank you for the beautiful stack trace :)
connection state seems ok, but `rx_chain` is obviously corrupt. hm.
since the line number doesn't match up to current master, can you give me the exact revision of mongoose.c or just attach it here? thanks!

kzyapkov, 2016-12-30:
mongoose-iot is at 5ce6ae36

I just found the `debug_coredump` target which maps more stuff, so here's an even prettier one:

```
Loaded core dump from last snippet in  /console.log
bssl_socket_recv (ctx=0x3fff7cfc, buf=0x3fff7e4c "", len=5)
    at /mongoose-iot/mongoose/mongoose.c:14628
14628	  size_t seg_len = (seg->len - cs->rx_offset);
tadd symbol table from file "/mongoose-iot/common/platforms/esp8266/rom/rom.elf" at
	.text_addr = 0x40000000
Reading symbols from /mongoose-iot/common/platforms/esp8266/rom/rom.elf...(no debugging symbols found)...done.
 (gdb) bt full
#0  ssl_socket_recv (ctx=0x3fff7cfc, buf=0x3fff7e4c "", len=5)
    at /mongoose-iot/mongoose/mongoose.c:14628
        nc = 0x3fff7cfc
        cs = 0x3fff5e64
        seg = 0x800
        __func__ = "ssl_socket_recv"
        seg_len = <optimized out>
#1  0x402449c3 in mbedtls_ssl_fetch_input (ssl=ssl@entry=0x3fff7c04, 
    nb_want=nb_want@entry=5) at ssl_tls.c:2513
        ret = <optimized out>
        len = 5
        want_buf_size = 13
#2  0x40245e70 in mbedtls_ssl_read_record_layer (ssl=ssl@entry=0x3fff7c04)
    at ssl_tls.c:3920
        ret = <optimized out>
#3  0x402466e6 in mbedtls_ssl_read_record (ssl=ssl@entry=0x3fff7c04)
    at ssl_tls.c:3865
        ret = <optimized out>
#4  0x402484d5 in ssl_parse_server_hello (ssl=0x3fff7c04) at ssl_cli.c:1456
        buf = <optimized out>
        t = <optimized out>
        ret = <optimized out>
        n = <optimized out>
---Type <return> to continue, or q <return> to quit---
        ext_len = <optimized out>
        comp = <optimized out>
        i = <optimized out>
        ext = <optimized out>
        handshake_failure = 0
        suite_info = <optimized out>
#5  mbedtls_ssl_handshake_client_step (ssl=ssl@entry=0x3fff7c04)
    at ssl_cli.c:3416
        ret = <optimized out>
#6  0x40244d5c in mbedtls_ssl_handshake_step (ssl=ssl@entry=0x3fff7c04)
    at ssl_tls.c:6470
        ret = -28800
#7  0x40244dc4 in mbedtls_ssl_handshake (ssl=0x3fff7c04) at ssl_tls.c:6494
        ret = 0
#8  0x4021e2be in mg_ssl_if_handshake (nc=nc@entry=0x3fff7cfc)
    at /mongoose-iot/mongoose/mongoose.c:4490
        ctx = 0x3fff5e84
        err = <optimized out>
#9  0x40225014 in mg_lwip_ssl_do_hs (nc=nc@entry=0x3fff7cfc)
    at /mongoose-iot/mongoose/mongoose.c:14478
        cs = 0x3fff5e64
        server_side = <optimized out>
        res = <optimized out>
---Type <return> to continue, or q <return> to quit---
        __func__ = "mg_lwip_ssl_do_hs"
#10 0x402252c8 in mg_ev_mgr_lwip_process_signals (
    mgr=mgr@entry=0x3fff4550 <s_mgr>)
    at /mongoose-iot/mongoose/mongoose.c:14287
        nc = 0x3fff7cfc
        cs = 0x3fff5e64
        md = 0x3fff50dc
#11 0x402254f5 in mg_lwip_if_poll (iface=<optimized out>, 
    timeout_ms=<optimized out>) at /mongoose-iot/mongoose/mongoose.c:14353
        mgr = 0x3fff4550 <s_mgr>
        n = 0
        nc = <optimized out>
        tmp = <optimized out>
        min_timer = 0
        num_timers = 0
#12 0x4021dca2 in mg_mgr_poll (m=m@entry=0x3fff4550 <s_mgr>, 
    timeout_ms=timeout_ms@entry=0) at /mongoose-iot/mongoose/mongoose.c:2279
        i = 0
        now = <optimized out>
        __func__ = "mg_mgr_poll"
#13 0x40219d8a in mongoose_poll (ms=ms@entry=0)
    at /mongoose-iot/fw/src/miot_mongoose.c:52
        ret = <optimized out>
---Type <return> to continue, or q <return> to quit---
        __func__ = "mongoose_poll"
#14 0x40226c8b in miot_lwip_task (e=<optimized out>)
    at /mongoose-iot/fw/platforms/esp8266/user/esp_task.c:82
        timeout_ms = <optimized out>
        mgr = 0x3fff4550 <s_mgr>
#15 0x40000f49 in ets_post ()
No symbol table info available.
Unmapped addr 0x2024f54b
#16 0x2024f54b in ?? ()
No symbol table info available.
Unmapped addr 0x2024f54b
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

rojer, 2016-12-30:
i cannot find 5ce6ae36 among the commits, and neither does github (since it's not linkified)

rojer, 2016-12-30:
also, is this easily reproducible in your setup? if yes, please raise `debug.level` to 3 and set `debug.mbedtls_level` to 3 as well and post the log. i'd like to see how far in the handshake it gets.

kzyapkov, 2016-12-30:
Sorry about that -- I forgot that we build from our fork, with a bunch of minor fixes integrated (we had pending issues for most if not all).

Here it is:
https://github.com/ALLTERCO/mongoose-iot/tree/allterco_fixes

There is no clear path for reproducing the issue, it happens sporadically on WiFi reconnect. I'll follow up when it occurs again with the verbose logs.

kzyapkov, 2017-01-17:
While trying to reproduce this we stumbled upon a significant number of very weird-looking processor exceptions. Many times heap integrity was broken, frequently a bunch of global variables got overwritten, etc.

I'm closing this one and will try to report some more concise results.

closed 2017-01-17 by kzyapkov

rojer, 2017-01-17:
fwiw, i did some testing downloading a file from the device with SSL on, in both station and AP mode, client and server, connecting and disconnecting and have not encountered this.

kzyapkov, 2017-01-17:
For us, 99% of the time we find corrupt memory or bad heap integrity immediately after wifi reconnect. May be some of our code causing the trouble.

rojer, 2017-01-17:
there are currently no known bugs in mgos that manifest like that (which is not to say there aren't any, of course).

heap integrity error is usually a use-after-free. integrity is checked on every malloc/free, meaning it must've happened fairly close to the place where it was detected. unfortunately, i don't have any suggestions for narrowing it down other than inserting printfs in a bunch of places...