cesanta/mongoose-os #177: miot tool doesn't run against master.

MobiusHorizons, 2017-01-01:
The `miot` tool necessary to build projects available from your website can't run against the `master` branch of this repository because of two commits:  19a62f81cada85c1e560636d8522a34e21cc4bfe , and cc559672b75b154bcbe86737a4149b53c90e7f73. Please either update the tool for download or provide documentation about which commit/branch can be used for development.

Thanks.

cpq, 2017-01-01:
Thanks for reporting, working on it now.

Could you elaborate on how do you plan to use Mongoose OS , please?

cpq, 2017-01-01:
Fixed, and verified manually on macos - please verify and confirm.
Download `mgos` tool from https://mongoose-iot.com/software.html.

`mgos build --arch ...`

closed 2017-01-01 by rojer

MobiusHorizons, 2017-01-01:
Great, I'll test it.

For the time being I am testing out mongoose-os as a possible base for some of the projects i had been working on, since it works on the boards i was using, and has better `mqtt` (with ssl) support than other platforms out of the box. 

It looks very promising.

MobiusHorizons, 2017-01-01:
verified and confirmed.