cesanta/mongoose-os #178: aws mqtt can't connect

MobiusHorizons, 2017-01-01:
When running the `c_mqtt` example on either the esp8266 or the esp32, the board will connect, but disconnects immediately with no CONNACK. A trace of the log is provided.

```
mqtt_global_connect  MQTT connecting to xxxxxxxxxxxxxx.iot.us-east-1.amazonaws.com:8883
SW ECDSA verify curve 3 hash_len 32 sig_len 70
SW ECDSA verify curve 3 hash_len 32 sig_len 71
ATCA is not available (-20096), using sw ECDH
ev_handler           MQTT Connect (1)
ev_handler           MQTT Disconnect
mqtt_global_connect  MQTT connecting to xxxxxxxxxxxxxx.iot.us-east-1.amazonaws.com:8883
SW ECDSA verify curve 3 hash_len 32 sig_len 70
SW ECDSA verify curve 3 hash_len 32 sig_len 70
ATCA is not available (-20096), using sw ECDH
ev_handler           MQTT Connect (1)
ev_handler           MQTT Disconnect
```

This is run with the base NodeMCU board for esp8266, and the Adafruit clone of the DevkitC for the esp32. 

Am I doing something wrong, or is this a bug?
Thanks for building this interesting product.

cpq, 2017-01-01:
Does c_mqtt connect to the default MQTT server (which is not AWS IoT) ?

MobiusHorizons, 2017-01-01:
The board is set up using the `mgos aws-iot-setup` process, and is connectiong to `xxxxxxxxxxxxxx.iot.us-east-1.amazonaws.com:8883` where `xxxxxxx` is the url created for me by aws.

cpq, 2017-01-01:
AWS MQTT server disconnects if there are some problems with the policy.
Please test c_mqtt example as the first part of README suggests - using a public MQTT server, that will help to pinpoint the problem.

MobiusHorizons, 2017-01-01:
That was the problem. Is there any chance that an error message could be shown such as `connection refused`

cpq, 2017-01-01:
Well, technically it's not refused. It is perfectly successful, but AWS silently closes it soon if the policy is not right.

MobiusHorizons, 2017-01-02:
ok, that makes sense. Thanks for your help.

MobiusHorizons, 2017-01-09:
I am fine with closing this issue, should I go ahead and do that?

cpq, 2017-01-09:
Thanks Paul!

closed 2017-01-09 by cpq