cesanta/mongoose-os #179: stm32 can't flash

MobiusHorizons, 2017-01-04:
`mgos` seems to be concatenating the port name and the file it is attempting to write, leading to the following error:

```
Loaded c_hello/stm32 version 1.0 (20170104-042207/???)
Copying c_hello.bin to /dev/tty.wchusbserial1420...
Error: open /dev/tty.wchusbserial1420/c_hello.bin: not a directory
```

cpq, 2017-01-04:
@rojer please take a look

MobiusHorizons, 2017-01-10:
is the source for the `mgos` tool somewhere public? if so, I could take a look for the error myself.

cpq, 2017-01-10:
Are you trying to flash STM32 board?

MobiusHorizons, 2017-01-10:
Yes, I build the `c_hello` project for `arch stm32`, and I was trying to flash to the serial bootloader.

cpq, 2017-01-12:
Got it, thank you @MobiusHorizons !
STM32 is not yet functional, work in progress!

cpq, 2017-02-07:
Fixed, please verify.

closed 2017-02-07 by cpq