cesanta/mongoose-os #180: Inconsistent change from "miot" to "mgos"

gicho, 2017-01-04:
Current (@today) version of "mgos" tool does not work for arch esp8266. 
- **mgos init** generates sceleton with "miot.yml" file name (should be "mgos.yml" probably?)
- **mgos build --arch esp8266** fails with "Error: open mgos.yml: The system cannot find the file specified."
- renaming "miot.yml" to "mgos.yml" allow the build to start but leads to compilation error:
`  CC    /mongoose-os/fw/platforms/esp8266/user/esp_config.c -> /app/build/objs/esp_config.o
src/main.c:1:29: fatal error: fw/src/miot_app.h: No such file or directory
 #include "fw/src/miot_app.h"
                             ^
compilation terminated.`

cpq, 2017-01-04:
Fixed. Please verify & confirm.

closed 2017-01-04 by cpq

gicho, 2017-01-04:
Yep, all good. Thank you.