cesanta/mongoose-os #181: mgos config-set esp32

blankcheck101010, 2017-01-12:
I am following the quickstart guide on the main page, and would eventually like to get c_mqtt working with AWS IOT. When running the code below

mgos config-set wifi.ap.enable=false wifi.sta.enable=true \
  wifi.sta.ssid=WIFI_NETWORK_NAME wifi.sta.pass=WIFI_PASSWORD

the device gets hung at

Getting configuration...
Setting new configuration...

^C

I eventually have to ^c out of it, and when I do, I try running mgos config-get and see no changes.

Same thing happens with mgos aws-iot-setup --aws-iot-policy myPolicy

Wrote private key to aws-iot-8c76c158f7.key.pem
Wrote certificate to aws-iot-8c76c158f7.crt.pem
Attaching policy "myPolicy" to the certificate...
Uploading certificate...

Then it freezes right there. Any idea why this is happening?

MobiusHorizons, 2017-01-12:
I had this problem on the esp8266 sometimes. In my case, I was able to
resolve it by restarting the device shortly before running the first
`config-set`, after which the other commands seemed to work.

blankcheck101010, 2017-01-12:
Thanks for the feedback. At least I know I'm not the only one. I've tried on the esp8266 and esp32. I've tried your combination, and it gets a little farther...

Getting configuration...
Setting new configuration...
Saving and rebooting...

but still hangs... weird. Have you got this working with the new mgos? I've only seen this work with miot

MobiusHorizons, 2017-01-12:
I have had it working for both esp32 (Adafruit devkit) and esp8266(nodemcu)
with mgos

blankcheck101010, 2017-01-12:
noticed a new software version this morning... just updated and so far the config-set command completed

blankcheck101010, 2017-01-12:
Just tested the c_mqtt example and the aws-iot-setup command also completed. Looks like they fixed this issue on the most recent version.

Sauls-MacBook-Pro:c_mqtt Saul$ mgos --version
The Mongoose OS command line tool
Version: 2017011212
Build ID: 20170112-121055/lsm@38aa3e1e+
Sauls-MacBook-Pro:c_mqtt Saul$

Thanks for chiming in MobiusHorizons and for the people in the background who updated the software ;)

MobiusHorizons, 2017-01-12:
Glad you were able to get it working

closed 2017-01-13 by blankcheck101010