cesanta/mongoose-os #182: esp32 can't flash

aifer2007, 2017-01-16:
$ mgos init --arch esp32
Connecting to http://mongoose.cloud, user  ...
Downloading project skeleton...
Unpacking...
Setting arch "esp32"...

$ mgos build
Connecting to http://mongoose.cloud, user test
Uploading sources (1629 bytes)
Success, built mgos32/esp32 version 1.0 (20170116-123239/???).
Firmware saved to build/fw.zip

$ mgos flash
Loaded mgos32/esp32 version 1.0 (20170116-123239/???)
Opening /dev/tty.SLAB_USBtoUART...
Connecting to ESP32 ROM, attempt 1 of 10...
  Connected
Running flasher @ 460800...
  Flasher is running
Flash size: 4194304, params: 0x0220
Error: nvs: failed to get data: "nvs": no source specified

$

aifer2007, 2017-01-16:
When I use "--arch esp8266", I get errors as:

$ mgos init --arch esp8266
Connecting to http://mongoose.cloud, user  ...
Downloading project skeleton...
Unpacking...
Setting arch "esp8266"...

$ mgos build
Connecting to http://mongoose.cloud, user test
Uploading sources (1631 bytes)
Success, built mgos32/esp8266 version 1.0 (20170116-123115/???).
Firmware saved to build/fw.zip

$ mgos flash
Loaded mgos32/esp8266 version 1.0 (20170116-123115/???)
Opening /dev/tty.SLAB_USBtoUART...
Connecting to ESP8266 ROM, attempt 1 of 10...
  Connected
Running flasher @ 460800...
Error: failed to run flasher: failed to run flasher stub: failed to write params: failed to start mem write: command 5 failure: error code 0

$ mgos flash
Loaded mgos32/esp8266 version 1.0 (20170116-123115/???)
Opening /dev/tty.SLAB_USBtoUART...
Connecting to ESP8266 ROM, attempt 1 of 10...
  Connected
Running flasher @ 460800...
Error: failed to run flasher: failed to run flasher stub: failed to write params: failed to start mem write: command 5 failure: error code 0

rojer, 2017-01-16:
```
Error: nvs: failed to get data: "nvs": no source specified
```

please update your `mgos` tool, this is a recent change

closed 2017-01-16 by rojer

aifer2007, 2017-01-17:
Thank you. It works well.