cesanta/mongoose-os #185: Can mos load a NPM module, such a Socket.io.min.js?

aifer2007, 2017-02-07:
Can mos load a NPM module, such a Socket.io.min.js?

cpq, 2017-02-07:
The mJS engine that is used by default, is very restricted and cannot use any node module.
The socketio lib is too fat to run on most of the boards that mOS targets, anyway.

closed 2017-02-07 by cpq

aifer2007, 2017-02-07:
Thks!

But how can I connect a mos device to a server running socket.io protocol?

cpq, 2017-02-07:
What exactly protocol socket.io is using?
Could you give more details about the project you're working on, please?

MobiusHorizons, 2017-02-07:
@aifer2007 you may wish to not use websockets, since they are just an expensive version of normal sockets built on top of http to allow web browsers to do two way communication. They exist for security purposes because if browsers could do arbitrary normal socket connections from javascript, they could hack a private network when a user just opens a page. To get the other benefits of Socket.io, you may wish to look into the third bullet point of http://socket.io/blog/introducing-socket-io-1-0/#new-engine, which would allow for Socket.io over normal TCP sockets. 

That said, there are websockets libraries that are small enough to run on embedded boards, but it is an unnecessary overhead.

cpq, 2017-02-07:
Thanks @MobiusHorizons 
Mongoose OS has a built-in networking library, https://github.com/cesanta/mongoose
Which has Websocket support.