cesanta/mongoose-os #186: Crash in mg_lwip_if_udp_send()

kzyapkov, 2017-02-09:
mongoose.c, line 13979, the pointer returned from `pbuf_alloc` is not checked before being used.

This causes crashes with mDNS enabled, in combination with [this issue](https://forum.cesanta.com/index.php?p=/discussion/213/ping-pong-basic-dns-sd-over-mdns)

```
  struct pbuf *p = pbuf_alloc(PBUF_TRANSPORT, len, PBUF_RAM);
  ip_addr_t *ip = (ip_addr_t *) &nc->sa.sin.sin_addr.s_addr;
  u16_t port = ntohs(nc->sa.sin.sin_port);
  memcpy(p->payload, buf, len);
```

One instance of a crash:

```
(gdb) bt   
#0  mg_lwip_if_udp_send (nc=0x3fff045c, buf=0x3fffbda4, len=223)
    at /mongoose-os/mongoose/mongoose.c:13982
#1  0x40215cd8 in mg_send (nc=nc@entry=0x3fff045c, buf=0x3fffbda4, len=223)
    at /mongoose-os/mongoose/mongoose.c:2457
#2  0x40218958 in mg_dns_send_reply (nc=nc@entry=0x3fff045c, 
    r=r@entry=0x3ffff568) at /mongoose-os/mongoose/mongoose.c:10370
#3  0x4021ce9f in handler (nc=0x3fff14fc, ev=<optimized out>, 
    ev_data=0x3ffff5f0) at /mongoose-os/fw/src/mgos_dns_sd.c:341
#4  0x40223335 in handler (nc=0x3fff14fc, ev=100, ev_data=0x3ffff5f0)
    at /mongoose-os/fw/src/mgos_mdns.c:76

...

(gdb) print p
$4 = (struct pbuf *) 0x0
```

rojer, 2017-02-10:
ok, so the immediate issue is fixed - in case of an OOM, the packet will just be dropped.
however, i wonder why you are hitting this at all. are you running close to memory limit?

on the ping-pong issue - that has been fixed, you should stop seeing it now.

kzyapkov, 2017-02-10:
This wasn't easily reproducible and has not occurred since we deployed the ping-pong fix. Definitely memory exhaustion here, debug log shows `malloc()` failing.

closed 2017-02-10 by kzyapkov