cesanta/mongoose-os #187: Allow setting DHCP hostname

kzyapkov, 2017-02-09:
This is for ESP8266, but probably applies to other platforms as well. The ESP SDK provides `wifi_station_set_hostname(char*)`, which does the trick. Maybe a single `hostname` config key can be used for this purpose as well as default mDNS hostname, etc.

I see `["device.id", "esp8266_??????"],` in `esp8266/src/esp_sys_config.yaml` but it doesn't seem to be used anywhere.

cpq, 2017-02-09:
I think the config setting `wifi.ap.ssid` is what you want.
please confirm.

kzyapkov, 2017-02-09:
Nope, what you mention is the SSID string for AP mode of the WiFi interface.

I am talking about DHCP request option 12 in the DHCP request sent by the device when in STA mode. Many routers use this option to set the hostname of the device in the local resolver.

cpq, 2017-02-09:
got it thanks @kzyapkov 
@alashkin please add

kzyapkov, 2017-02-13:
Sadly, it seems this option is not supported by CC3200, at least with the current NWP firmware:

https://e2e.ti.com/support/wireless_connectivity/simplelink_wifi_cc31xx_cc32xx/f/968/p/420212/1498318#1498318
http://e2e.ti.com/support/wireless_connectivity/simplelink_wifi_cc3000/f/851/p/350167/1229570#1229570

I have no experience with the other supported platforms.

kzyapkov, 2017-02-27:
Bumping this, since it really looks trivial to impelment for ESP8266. All that's needed is for wifi setup code to call `wifi_station_set_hostname()` before `wifi_station_connect()`. Alas, it doesn't work if we call it outside of `mgos_wifi_setup_sta().` Also, there's the ESP8266-specific `device.id` config key ...

alashkin, 2017-03-23:
Implemented in 57948f396e5098f2cc9249e0103e694056fa9ea4

closed 2017-03-23 by alashkin

rojer, 2017-03-23:
also, CC3200 SDK and NWP update is on the way and it allows setting dhcp hostname. once that lands, i'll update mOS to do it.

rojer, 2017-03-23:
hm. so, latest NWP update has this item in the changelog:
```
SL_WIFI_R1-1617	DHCP: Add support for hostname field in dhcp request
```
so NWP can do it now, but i don't see any new API calls or options in the host driver in SDK 1.3.0.
[posted a question on E2E](https://e2e.ti.com/support/wireless_connectivity/simplelink_wifi_cc31xx_cc32xx/f/968/t/583491).

hrshtmims, 2018-04-29:
@kzyapkov Hostname is set, but I am unable to connect via hostname. Hostname is reflecting in Router's clientlist.

kzyapkov, 2018-04-30:
I don't think this ever happened on CC32xx, but works fine for me on espressif chips. Check your local resolver with `dig` or `nslookup`, make sure you are using the correct tlv.