cesanta/mongoose-os #188: Ability to enable COAP

kzyapkov, 2017-02-13:
With the new `mos` tool, there doesn't seem to be a way for setting `MG_ENABLE_COAP` when building the mongoose networking library.

Additionally, we override a few other mongoose flags. Maybe some are not needed anymore, but please take a look:

```
    -DMG_DISABLE_HTTP_KEEP_ALIVE \
    -DMG_DISABLE_DAV_AUTH \
    -DMG_DISABLE_DAV \
    -DMG_ENABLE_HTTP_STREAMING_MULTIPART \
    -DMG_LOCALS \
    -DMG_FS_SPIFFS \
    -DMG_ENABLE_FILESYSTEM \
    -DMG_ENABLE_SSL \
    -DMG_ENABLE_COAP \
    -DMG_ENABLE_DIRECTORY_LISTING
```

cpq, 2017-02-17:
Setting through build_vars should work:

https://mongoose-os.com/docs/#/quickstart/build.md/

See example at 
https://github.com/cesanta/mongoose-os/blob/master/fw/examples/c_mqtt/mos.yml#L10-L11

Please confirm it works as expected and close this.

kzyapkov, 2017-02-27:
Setting it through `build_vars` works, thanks!

Offtopic, we chose CoAP because it's a standard, and hoped to make it somewhat easier for others to integrate with our devices. It works OK for status broadcasts, device-to-device signalling and basic commands. Why didn't you like it?

cpq, 2017-02-27:
@kzyapkov cause CoAP is complex. It's ok for smallish things, but when it comes to big messages - fragmentation comes into play (reimplementation of TCP), and when it comes to security - DTLS which is a pain.
In my opinion, only a quite narrow subset of vendors promote CoAP.

closed 2017-02-27 by cpq