cesanta/mongoose-os #189: Sparkfun ESP32 Thing & Adafruit Feather Huzzah not supported

sh4nnongoh, 2017-02-17:
Unable to connect to the wifi during the setup wizard.
Are these boards supported?

sh4nnongoh, 2017-02-23:
Ok it is supported, got it working using the cmdline.
Note that the UI Wifi setup does not work.
Thanks.

cpq, 2017-02-23:
Are you using windows?

sh4nnongoh, 2017-09-22:
Hi reopening this.

Yes I am using windows.

The Mos UI flashing process fails when performing it in the Web UI.

The commandline works for Adafruit Feather Huzzah but fails for **Sparkfun ESP32 Thing**.

[image]

cpq, 2017-09-25:
Could you confirm that with the latest `mos` build please?

`mos update latest`

sh4nnongoh, 2017-09-29:
Hi, refer to the images below.
I am using **mos build 20170922-103145/master@e973ef66+** on Windows.

# SparkFun ESP32 Flash Fail through UI
[image]

# SparkFun ESP32 Flash Fail through CLI
[image]

# Adafruit Huzzah Flash Fail through UI
[image]

# Adafrauit Huzzah Flash Pass through CLI (**After multiple attempts**)
[image]

dgdosen, 2018-02-06:
Any update to this? Can we load mongoose on an ESP32 Thing?

cpq, 2018-02-07:
A multiple relevant updates to the `mos` tool has been made since - please try and report back.

mike-rogers, 2018-02-11:
@dgdosen - i have been able to load mongoose os firmware on an ESP32 thing without issue. i'm using the ESP32 to drive an AWS IoT-enabled gong, much to the delight of my coworkers.

@cpq - i just now updated `mos` and while `mos flash` works with the ESP8266-based Feather Huzzah I am unable to use the `mos wifi` command successfully:

```
Using port /dev/cu.SLAB_USBtoUART
Getting configuration...
Error: write /dev/cu.SLAB_USBtoUART: file already closed
```

the output of `mos --version`:
```
The Mongoose OS command line tool
Version: 1.26
Build ID: 20180208-184624/1.26@ab7f50e1+
Update channel: release
```

this is all on mac. if you'd like any more information please let me know. :+1:

cpq, 2018-02-13:
@mike-rogers is that intermittent or happens all the time? what `mos console` say?

mike-rogers, 2018-02-13:
It was consistent, but in trying to get you `mos console` output I may have fried my only ESP8266. ☹️

rohit2787, 2018-07-27:
@cpq I am getting a similar error. I am trying to use mos to flash esp32. And I have successfully used both command line and webUI with ESP32 successfully for the past several months. But now i experience an error where when I open the webUI, and select the connection, the flashed code does not load. The webUI gets frozen and I have to close it. I can see the output of my code on mos console. But not the webUI. Also, I noticed that when I load an older code - which used to work, the web UI starts working. But my current code does not. Is there an issue with using for loop in JS?

mike-rogers, 2019-04-29:
A coworker handed me a Feather HUZZAH32 out of the blue and I was unable to use `mos wifi` through either the command line nor the mos UI - imagine my surprise to see my own face after googling for a few minutes! Needless to say this is still broken - but `mos console` and `mos flash` still work both through the command line and the mos UI.

```
Using port /dev/cu.SLAB_USBtoUART
Getting configuration...
Error: write /dev/cu.SLAB_USBtoUART: file already closed
/private/tmp/mos-20190426-57195-kki69o/mos-tool-2034cc00c37fee33b300b1c5869b8eeff328fdfa/src/cesanta.com/common/go/mgrpc/codec/serial.go:187: 
/private/tmp/mos-20190426-57195-kki69o/mos-tool-2034cc00c37fee33b300b1c5869b8eeff328fdfa/src/cesanta.com/common/go/mgrpc/codec/serial.go:246: 
/private/tmp/mos-20190426-57195-kki69o/mos-tool-2034cc00c37fee33b300b1c5869b8eeff328fdfa/src/cesanta.com/common/go/mgrpc/codec/stream.go:315: 
/private/tmp/mos-20190426-57195-kki69o/mos-tool-2034cc00c37fee33b300b1c5869b8eeff328fdfa/src/cesanta.com/common/go/mgrpc/mgrpc.go:383: 
/private/tmp/mos-20190426-57195-kki69o/mos-tool-2034cc00c37fee33b300b1c5869b8eeff328fdfa/src/cesanta.com/mos/dev/dev_conn_impl.go:152: 
/private/tmp/mos-20190426-57195-kki69o/mos-tool-2034cc00c37fee33b300b1c5869b8eeff328fdfa/src/cesanta.com/mos/dev/dev_conn_impl.go:175: 
/private/tmp/mos-20190426-57195-kki69o/mos-tool-2034cc00c37fee33b300b1c5869b8eeff328fdfa/src/cesanta.com/mos/dev/dev_conn.go:101: 
/private/tmp/mos-20190426-57195-kki69o/mos-tool-2034cc00c37fee33b300b1c5869b8eeff328fdfa/src/cesanta.com/mos/config/config.go:70: 
/private/tmp/mos-20190426-57195-kki69o/mos-tool-2034cc00c37fee33b300b1c5869b8eeff328fdfa/src/cesanta.com/mos/main.go:178: wifi failed
```

mike-rogers, 2019-04-29:
I was able to get `mos wifi` to work by making my `mos.yml` look like [this example](https://github.com/mongoose-os-apps/wifi-setup-web/blob/master/mos.yml), with special attention to the `libs` section.