cesanta/mongoose-os #190: Request: Actions on Google (for Google Assistant)

Hedda, 2017-02-17:
Please consider adding support for "Actions on Google" (for Google Assistant, including Google Home Speaker)

https://developers.google.com/actions/

Actions on Google let you build for the Google Assistant (which is Google's version of Amazon Alexa). Google lets developers create “conversation actions” which acts like little bots that users are able to interact with using the Google Home speaker. Actions are a lot like Skills for Amazon's Alexa speakers in that they allow third-party developers to easily integrate their apps and services with the brains of the Google Home speakers.

With the Google Assistant’s SDK, developers are be able to create two kinds of actions – Direct and Conversation. The former entails simple requests like home automation, while the latter is more of a back and forth interaction utilizing API.ai

API.AI makes it possible for your service to receive and respond to user queries in natural language. 

https://api.ai/blog/2016/12/08/build-conversation-actions-for-the-google-assistant/

The open system will also make it possible to utilize Assistant on third-party hardware, from home-brewed devices to large-scale consumer electronics. Interested parties can sign up for the service and updates through Google

https://services.google.com/fb/forms/6c9aa7f0641411e6a3a3a9197c7cdd89-copy-copy/

With support for "Actions on Google" you can control IoT just like you can with Amazon Echo.

https://madeby.google.com/home/

Google Home is a voice-activated speaker powered by the Google Assistant. Actions on Google allow you to add control of third-party services.

https://www.cnet.com/how-to/how-to-use-third-party-actions-on-google-home/

https://www.infoq.com/news/2016/12/Google-AI-API

http://www.theverge.com/2016/12/8/13878444/google-home-developers-actions-ecosystem-app-store

cpq, 2017-02-17:
Google is trying to catch up with Amazon, eh ? :)

Could you give a good PRACTICAL example for the actions integration please?
Do you personally work on any related project?

Hedda, 2017-02-17:
Yes Google is trying to catch up with Amazon, with Google's “Conversation Actions" (a.k.a. 'Assistant Actions') being the same as Amazon's "Alexa Skills", and "API.ai" being the same as "Amazon AWS IoT" services. So I really anything that applies to Amazon AWS IoT could be setup for https://api.ai to enable Google Actions.

I guess practical example could be for Mongoose to have its own service on API.ai and build actions for common IoT tasks such as 'Turn ON Lights', 'Turn OFF Lights', 'Dim Lights', and 'Turn Up Lights', which would then allow users to use their Google Home speakers (and Google Assistant on smart phones/tablets) to control their home automation controlled by connected ESP8266/ESP32 to power relays.

https://api.ai/blog/2016/12/08/build-conversation-actions-for-the-google-assistant/

No sorry I have no personally work on any related project.

Hedda, 2017-02-17:
Checkout https://docs.api.ai/docs/actions-on-google-integration and

you can test with Alexa as https://docs.api.ai/docs/alexa-integration

cpq, 2017-04-09:
I believe you're talking about similar:
https://twitter.com/i/moments/849254058347122688

For now, Mongoose OS has enough on its plate to implement on a lower level.
So closing this, and leaving actions/alexa to the community.

closed 2017-04-09 by cpq