cesanta/mongoose-os #191: mos has no javascript UART and SPI APIs? and RPCs?

aifer2007, 2017-02-19:
mos has no javascript UART and SPI APIs? and RPCs?

aifer2007, 2017-02-19:
I make my own uart js api  for ESP32.

api_uart.js:
let Serial = {
  //
  setStdout: ffi('int mgos_set_stdout_uart(int)'),
  setDebug: ffi('int mgos_init_debug_uart(int)'),
  write: ffi('int mgos_uart_write(int, char *, int)'),
  enableWrite: ffi('void mgos_uart_set_write_enabled(int, bool)'),
};

in mos.yml: 
ffi_symbols:
  - mgos_uart_write
  - mgos_init_debug_uart
  - mgos_uart_init
  - mgos_uart_is_inited

and then, in init.js:
// Load Mongoose OS API
load('api_timer.js');
load('api_gpio.js');
load('api_uart.js');

let uartNo = 1;
Serial.setDebug(uartNo); // baut: 115200
let str = "Hello from ESP32-Serial: " + JSON.stringify(uartNo) + "\r\n";
let isInited = Serial.isInited(uartNo);

// Blink built-in LED every second
let PIN = ffi('int get_led_gpio_pin()')();  // Helper C function that returns a
                                            // built-in LED GPIO
GPIO.set_mode(PIN, GPIO.MODE_OUTPUT);
Timer.set(1000 /* milliseconds */, 1 /* repeat */, function(pin) {
  let value = GPIO.toggle(pin);
  print(value ? 'Tick' : 'Tock');
  print(isInited);
  print(Serial.write(uartNo, str, str.length));
}, PIN);

It works well. 

I can receive the string on my notebook with a UART-USB converter(FT232)。

[image]

aifer2007, 2017-02-19:
on ESP32, the UART2 is not available yet. 

And the pins of a UART can't be assigned by programmer.

cpq, 2017-03-02:
RPC api has been added

https://mongoose-os.com/video-tutorials.html#video6

cpq, 2017-03-24:
Working on UART API now.

aifer2007, 2017-03-28:
That's great!

cpq, 2017-04-09:
Done.
https://mongoose-os.com/docs/#/js-api/api_uart.js/

closed 2017-04-09 by cpq

aifer2007, 2017-04-09:
Great！

发自我的 iPhone