cesanta/mongoose-os #193: Problem with JSON parameter at MOS.EXE command line

Denis2106, 2017-02-19:
I try command > MOS.EXE CALL FS.Remove '{"filename": "test"}' and have response:
Error: Args ['{filename:test}'] is not a valid JSON string.

I try to write args: "{'filename':'test'}", {"filename": "test"}, {"filename": "test"}, {filename: test}  
In all cases i have one response: not valid JSON string.

MobiusHorizons, 2017-02-28:
escaped quotes might help`"{\"filename\":\"test\"}"`. You can always tell what's actually getting to the program by testing with echo. However on OSX `echo '{"filename":"test"}'` returns `{"filename":"test"}` as expected, but it's worth a shot anyway.

Uriziel01, 2017-02-28:
I figured it out that JSON strings are corrupted only in Windows default command line.
If You use another terminal like cygwin it works just fine.

It should easly resolved in mos.exe by parameters sanitization.

cpq, 2017-03-02:
@alashkin please fix that windows

alashkin, 2017-03-22:
Unfortunately, this behavior is normal for Windows: its command interpreter (aka command.com) uses double quote character for parameter concatenation. (ex: myprogram "param with space")
To get "real" quote you have to somehow escape double quotes.
ex:
`mos call FS.Remove {"""filename""":"""test"""}`
`mos call FS.Remove {\"filename\": \"test\"}`
Note, if you put space between {, filename, :, test or } you need to use " for grouping:
`mos call FS.Remove "{\"filename\":  \"test\" }"`

Not sure we can fix it in normal way, this is how windows work

closed 2017-03-22 by alashkin