cesanta/mongoose-os #198: Remote device management with unpredictable results

deanmaniatis, 2017-02-26:
An ESP8266 module has been commissioned with mongoose-os and while connected with USB all functionality seems to be working fine. Module gets untethered and use the following commands to communicate. Some work, some don't e.g the following don't work:

* console, 
* uploading an edited file and be able to see the updated content on HTTP server, e.g. I download index.html and change its content, I upload it on the module and I'm able to see from CLI and `mos get` that the content is updated, on HTTP server though the old content is still shown. I tried to restart the module with `mos Sys.Reboot` and didn't update content. 

http://paste.ubuntu.com/24071912/

cpq, 2017-02-26:
yes, at the moment console only works with serial.

w.r.t. uploaded file : that should work.
```
mos get <file>
...edit file...
mos put <file>
curl IP/<file>
```

deanmaniatis, 2017-02-26:
Hi,
I follow these exact steps, only i was testing with the browser. Indeed, curl shows the right file but still browser shows the old one (Firefox). I guess it is caching it although if i edit the file in mos UI and click the button that saves and restarts the board the page in the second tab get the updated value.

deanmaniatis, 2017-02-27:
@cpq Is `mos flash` also not available with remote protocol?
```zebook 福 ~/dev/mos/sample 
4170 ◯ : mos call RPC.Hello                                                                                                                                                     
{
  "time": 68591
}

4167 ◯ : mos flash                                                                                                                                                              
Loaded sample/esp8266 version 1.0 (20170227-100111/???)
Opening ws://192.168.2.31/rpc...
Error: failed to open control port: open ws://192.168.2.31/rpc: no such file or directory
```

cpq, 2017-02-27:
Yes. Flashing can only work, and will only work, with serial. 
Console now works with serial only, but potentially can run over the network too.

deanmaniatis, 2017-02-27:
@cpq In which way does "flashing" differ with pushing new updated firmware using the curl command?

closed 2017-02-27 by deanmaniatis

cpq, 2017-02-27:
Flashing talks directly to the boot loader via serial, wipes out everything, writes new firmware.
Flashing works when mOS is not yet installed.

curl command updates mOS over the air. That means mOS is already installed, and it can update itself.