cesanta/mongoose-os #199: Honor DNS server from DHCP

kzyapkov, 2017-02-27:
mos's DNS resolver implementation seems to explicitly use Google's 8.8.8.8 DNS server for lookups and doesn't honor DNS settings from DHCP. This doesn't work correctly in intranets or other environments where 8.8.8.8 is not accessible. Moreover, local resolving doesn't work.

Can you plaese fix this for CC3200 and ESP8266? Or, at least make it possible for devices to be configured to use the locally-provided DNS?

cpq, 2017-02-28:
@alashkin please implement

alashkin, 2017-03-21:
For esp32 & esp8266 implemented in https://github.com/cesanta/mongoose-os/commit/e77fa53040e69050d18f28d6f9bb007231c01a57

alashkin, 2017-03-21:
cc3200 implemented in https://github.com/cesanta/mongoose-os/commit/7a5b36dc27dc1c10c1912f71318a4ad015572c82

closed 2017-03-21 by alashkin

fx, 2018-04-14:
@alashkin is this in `2.0+4ed5853~xenial0`? Seems to ignore the dhcp nameserver still.