cesanta/mongoose-os #201: Helper c function that returns builtin LED error?

deanmaniatis, 2017-03-01:
Hello, got my hands on a new ESP32 dev. board and for some reason the `get_led_gpio_pin()` returns value 21 where in fact it should have returned 2. Is this something to do with esp32 firmware, mongoose API or something different?

cpq, 2017-03-01:
pin 21 for esp32  is correct.

The thing is , esp32 devkit C board does not have any built-in LED!
that's why we return just a random pin which is close to the GND pin, in order to use an external LED:

[image]

closed 2017-03-01 by cpq

deanmaniatis, 2017-03-02:
This is my board, designed by eboxmaker. The builtin LED is connected to PIN 2. 

[image]

deanmaniatis, 2017-03-02:
@cpq Is it worth implementing profiles for different boards ?

cpq, 2017-03-02:
Got it, thank you.
profiles, uhh... i think that'll be too much, to be honest..

`let PIN = 2;` would do the job :)